ra3xdh/qucs_s #96: Adding a filter generates lots of 'format errors'

darkstar007, 2022-07-02:
If I create I filter as show 
[image]
and then click 'calc and put into clipboard' and then paste into the main window,it generates lots of pop-up warnings as shown 
[image]
which have to be individually dismissed by clicking on them - its rather annoying and possibly broken?

Thanks,
Matt

ra3xdh, 2022-07-02:
The microstrip devices are SPICE-incompatible and set hidden if the Ngspice is selected as the primary simulation kernel. See also #94 The solution may be to hide the microstrip filters from the filter design tool if the SPICE mode is active to prevent such errors in the future.

darkstar007, 2022-07-02:
OK, thanks - setting the simulator to Qucsator and the issue go away.

I'd recommend greying the filters out (**and a tooltip to say why they are greyed out**) when ngspice is selected as the simulator would be a good approach.

Thanks,
Matt

ra3xdh, 2022-08-06:
Fixed on `current` branch. The incompatible filters are greyed now.

closed 2022-08-06 by ra3xdh