ra3xdh/qucs_s #103: Warning: QFSFileEngine::open: No file name specified

yurivict, 2022-07-08:
Qucs-S prints this warning several times during startup.

Version: 0.0.24
FreeNSD 13.1

ra3xdh, 2022-07-08:
Unfortunately I cannot reproduce this issue on Linux+Qt5.15. But @tvrusso has recently checked the program operation on FreeBSD and found no major problems. Probably he can help to debug the operation.

tvrusso, 2022-07-08:
I cannot reproduce this on my FreeBSD system on the "current" branch.  I have been building Qucs-s from source (manually, not via the port, and I don't use binary packages of ports) on BSD since the first Qt5-enabled version, and have never seen this startup message.

I just today built the master branch (a few minor commits past 0.0.24, at commit 76d1b6da ) and even that version does not print those warnings at all on startup.

tvrusso, 2022-07-08:
Then there is something fishy with how the port is built.

I built mine by cloning the source, and configuring with
```
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/some/path/in/my/home/directory
make install
```
and it has worked out of the box without this warning.

yurivict, 2022-07-08:
The port is built in a similar way.
There are no special options, it's built like all generic cmake/Qt projects.

tvrusso, 2022-07-08:
If you build it from source *outside* of the ports system, do you get the same behavior?

tvrusso, 2022-07-08:
I am currently in the middle of a big ports upgrade, so can't try out your port to see if I can reproduce this warning from the port-installed version.  It may be quite late today before I am able to do that.

But when I build from source without using the port, this doesn't happen.

yurivict, 2022-07-08:
In order to build a specific port from source you need to first run:
```sudo pkg install -A `make missing` ```

This is in case you are rebuilding all dependencies which can take a day.

tvrusso, 2022-07-08:
I do not use "pkg" to install binary packages, as I do not often like the options that the package builders have chosen.  I always source builds from ports (using make install).  I have been using FreeBSD for over 30 years (since before it was even called FreeBSD, but rather 386BSD), so am quite familiar with how to run the ports system.

yurivict, 2022-07-08:
I see. Good to know!

tvrusso, 2022-07-09:
I still cannot reproduce this on FreeBSD 12.3-STABLE.  I installed your port with `make install clean` from /usr/ports/cad/qucs-s (after removing the dependence on octave, which has dependencies I don't want on my system and which is NOT required for qucs-s to work in its basic capabilities), and fired it up.  There was no warning on startup.  I don't know where your warnings are coming from, but I don't get them.

(FWIW:  octave depends on arpack-ng, and the port maintainer of arpack-ng decided to add an unconditional dependence of arpack-ng on mpich, which octave doesn't even use.  mpich does not play well with openmpi, because the port for mpich dumps its libraries directly into /usr/local/lib and several of them have the same names as libraries installed by openmpi into /usr/local/mpi/openmpi/lib.  Programs linked with openmpi will fail to run if mpich is installed, because the mpich shared libraries in /usr/local/lib will be found and used instead of the openmpi shared libraries despite the RPATH/RUNPATH setting in the binaries.  And in my Xyce work I *require* openmpi.  Therefore, I cannot have octave installed.  Honestly, it doesn't make sense to make octave a required dependency since you can do plenty with qucs-s without having it.  It is just a nice add-on capability.)

yurivict, 2022-07-09:
Try ```rm -rf ~/.qucs/ ~/.config/qucs/``` and then run ```qucs-s```.

yurivict, 2022-07-10:
FYI: I took maintainership of arpack-ng a few years back but I didn't overhaul it yet because there was no pressing need. I'll look into it deeper now that you mentioned it.

I'll address your other concerns a bit later.

tvrusso, 2022-07-10:
Yep, removing the .qucs config makes it reproduce the warning.

That probably is enough to help @ra3xdh reproduce it on Linux, too.

ra3xdh, 2022-07-10:
Yes, I can reproduce it now. It comes from nonexistant config file `$HOME/config/qucs` when Qucs-S tries to read settings on the first startup. The issue should not affect the program operation. I will provide a fix soon.

> after removing the dependence on octave

The octave should not be a dependency for Qucs-S and required only for some specific applications. The only required dependencies for Qucs-S package are Qt5 libraries and Ngspice. Other tools (like ADMS, Octave) are optional.