ra3xdh/qucs_s #106: Pulse source doesn't generate signal

yurivict, 2022-07-08:
I am trying this example that contains a PULSE source:
[double-mosfet-example.sch.txt](https://github.com/Qucs/qucs-s/files/9074404/double-mosfet-example.sch.txt)

However, there is no pulse in the simulation:
[image]

Version: 0.0.24
FreeBSD 13.1

ra3xdh, 2022-07-08:
The pulse source by itself works as expected. There is a convergence issue with Ngspice. The simulator log shows "timestep too small" error.

yurivict, 2022-07-08:
This is strange because the example was taken almost verbatim from the ngspice [example on their page](https://ngspice.sourceforge.io/ngspice-tutorial.html)

Is Qucs-S able to save the project in SPICE format to run it outside Qucs-S?
I couldn't find the "Export as SPICE" menu item.

yurivict, 2022-07-08:
"Save Netlist" option exists on the Simulate page.

I'll report this to Ngspuce.

Thanks for your help!

closed 2022-07-08 by yurivict

yurivict, 2022-07-08:
https://sourceforge.net/p/ngspice/bugs/595/

ra3xdh, 2022-07-09:
Ngspice the most probably has no bug. You are using wrong transistor in this circuit. The BSP171 MOSFET is designed for 12V operation and for microsecond range. When I am replacing the power source and pulse duration the circuit begins to operate correctly. See the attached screenshot.
[image]