ra3xdh/qucs_s #108: When resistor's resistance is specified with space - ngspice fails

yurivict, 2022-07-09:
```6.8 k``` causes this failure:
```
Ngspice started...
warning, can't find model 'k' from line
    r1 whitenoise _net1 6.8 k
    Simulation interrupted due to error!
```

```6.8k``` works.

You might want to allow a single space optionally in the dialog box, but you should always remove it when writing SPICE.

ra3xdh, 2022-07-09:
What resistor are you using? There are two types of resistors in Qucs-S. Blue resistor is Qucs device and spaces are allowed in the resistance value for it. The red resistors use exact SPICE syntax. According SPICE specification the model may follow after the resistance value. So the spaces before unit for the red resistors are not allowed intentionally.

yurivict, 2022-07-09:
Red resistor is used.

closed 2022-08-13 by ra3xdh