ra3xdh/qucs_s #109: Warnings/failure in Xyce on simple circuit

yurivict, 2022-07-09:
Testcase:
[simple.sch.txt](https://github.com/ra3xdh/qucs_s/files/9077525/simple.sch.txt)
 
Xyce simulator prints these warnings:
```
***** Reading and parsing netlist...
Netlist warning in file /usr/home/yuri/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 11
 No model parameter VT0 found for model MMOD_M171 of type PMOS, parameter
 ignored.
Netlist warning in file /usr/home/yuri/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 13
 No model parameter VT0 found for model MMOD_M2 of type PMOS, parameter
 ignored.
Netlist warning in file /usr/home/yuri/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 15
 No model parameter VT0 found for model MMOD_M3 of type PMOS, parameter
 ignored.
Netlist warning in file /usr/home/yuri/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 28
 No model parameter VT0 found for model MMOD_M1 of type NMOS, parameter
 ignored.
```

Additionally Qucs-S reports these as warnings but the graph isn't updated. Was there really no data to report, or warnings were accidentally mishandled and prevented the graph update?

ra3xdh, 2022-07-09:
> Additionally Qucs-S reports these as warnings but the graph isn't updated. 

Yes, I see in the Xyce reference guide page 306 that it uses `VtO` instead of `Vt0`. This may be a netlister issue and requires additional convertion. 

> Additionally Qucs-S reports these as warnings but the graph isn't updated. 

It is not a bug. Qucs-S finishes the simulation and updates the graphs. Make sure that you have selected the appropriate simulator in the diagram properties window. You will see the `xyce:` prefix on traces that are related to Xyce.
[image]
[image]

ra3xdh, 2022-07-09:
The issue with `Vt0` and `VtO` comes from `qucs2spice.cpp` Qucs-S library devices are stored in legacy qucsator notation. I decide not to convert all existing libraries into SPICE notation while the design of the program. The Qucs-S doesn't convert the `Vt0` to `VtO` for the library devices. The system devices are not affected.

tvrusso, 2022-07-10:
For what it's worth, the Xyce team had completely forgotten about this SPICE incompatibility, and there's no reason anymore why we can't support VT0 as a synonym for VTO in the legacy (levels 1-6) MOSFET models -- this incompatibility has been around for as long as Xyce has existed.  I've started the process to get it into our development backlog in the hopes we'll have it fixed properly for our next release.

ra3xdh, 2022-07-10:
Fixed by https://github.com/ra3xdh/qucs_s/commit/60866f5a3d0167c9dc089d8cd39809b1827c4c24 Closing

closed 2022-07-10 by ra3xdh

tvrusso, 2022-07-10:
Just for what it's worth, vto is what SPICE2 used for this parameter.  SPICE2 didn't recognize vt0.

vt0 is recognized by SPICE3 and all its descendants because it was considered a typo that should just be recognized instead of erroring out.

Xyce doesn't recognize vt0 and never has, but that's just an oversight that we will probably correct very soon.  

There is a similar "o" vs. "0" thing for the parameter "uo" --- and Xyce has recognized "u0" as a "recognizable typo" since 2001 precisely to be compatible with SPICE3.