ra3xdh/qucs_s #111: Simulation doesn't work after the *.sch file is moved

yurivict, 2022-07-09:
When the file is moved to different folder/name - simulation doesn't show result any more.

It contains the old name in the file and this prevents simulation results from being displayed.

You probably need to update names in the file or better not store them at all.

ra3xdh, 2022-07-10:
Yes, it is a well known issue. The old Qt4 Qucs was also affected. The paths for `*.dat` and `*.dpl` are saved in the schematic file and are not updated when you make a copy using the file manager. Always use the `File->Save As` as the workaround. You can also input dataset names in the schematic properties dialog. As the solution I can propose to show a warning when the basename of the schematic and dataset are not matching after the open file. Something like "The name of schematic and dataset are not matching. Correct it automatically? Yes|No" would be helpful. 
[image]

ra3xdh, 2022-07-16:
I have added a fix by commit https://github.com/ra3xdh/qucs_s/commit/533180241eeeded7979e4e53c3223f029de566ca Now if the sch and dat/dpl file names are not matching the following message box will be shown. If the user press "Yes" the file names will be fixed automatically.
[image]

closed 2022-07-16 by ra3xdh