ra3xdh/qucs_s #113: [question] Do you plan to support .asc files?

yurivict, 2022-07-11:
The popular freeware circuit simulator LTspice saves .asc files.

Does Qucs-S plan to support such files?

ra3xdh, 2022-07-11:
The support of the `*.asc` schematic files is not considered, but the support of the LTSPICE library `*.asy` files format may help to use existing libraries instead of converting every library to Qucs `*.lib` format manually. The LTSPICE libraries have the plain text syntax, so in principle its support may be implemented.

yurivict, 2022-07-11:
Many people say that LTspice has many proprietary models that aren't available in ngspice.
Importing *.asy files will close this gap,

tomhajjar, 2022-07-12:
LTSpice *.asy files store the schematic symbol not the model parameters. 

The model or macromodel files have no fixed suffix. The suffix can be anything you want.  They generally conform to "standard" Spice formats. Proprietary model files can be encrypted and these cannot be opened  for viewing or modification. Anyone who needs to use proprietary models probably has access to LTSpice or commercial Spice simulators.  

Am I missing something?

ra3xdh, 2022-07-12:
The `*.asy` files contain the reference to the `*.lib` file that contains the plain text SPICE model. This reference could be extractd from `*.asy` file and used with `.include` directive. The models distributed with LTSPICE are not encrypted https://ltwiki.org/index.php?title=SPICE_Model_Links

ra3xdh, 2022-07-13:
I see the support of the `*.asy` files as following. The user can specify the directory containing the set of `asy` files and `lib` files. For example from LTSpice installation root. The Qucs can read these files, show in the system library manager and paste symbols on schematic using drag'n'drop.

ra3xdh, 2022-07-13:
> When you grab text to move it, a "ghost" rectangle appears at 0,0 regardless of the cursor location

Yes, I confirm bug with the texts dragging. As the workaround you can move the text with arrow keys on keyboard. This mode working as expected.

ra3xdh, 2022-07-13:
> Are you going to add more items to the nonlinear components library like in Qucs?

The 4-terminal BJT (red one SPICE modification) is also missing. All MOSFET devices have 4-pin notation in SPICE. I can add device with SB pins shorted internally.

tomhajjar, 2022-07-13:
I guess it depends on the model. A "basic" model uses 4 terminals and a "complex" one from the manufacturer can use 3. NMOSFETs.lib uses 3 terminals.

.SUBCKT FQT13N06 2 1 3 
;
;
M_BSIM3 16 6 7 7 Bsim3 W=0.451 L=2.0e-6 NRS=1
.MODEL Bsim3 NMOS (LEVEL=7....
;
;
.ENDS

ra3xdh, 2022-11-28:
Some questions regarding the LTSpice compatibility were raised in #141. LTSpice contain a lot of Ngspice-incompatible devices like mentioned opamp device. So the adding support of the `*.asc` files will be a little help for users. This extension will be not considered in the near future.

closed 2022-11-28 by ra3xdh