ra3xdh/qucs_s #116: Spice models with *.mod extension

tomhajjar, 2022-07-13:
I was wondering why Spice models with the "mod" extension show up in the "XSPICE" location of the Main dock. A number of Semi companies files use mod for the extension. 

[image]
"

tomhajjar, 2022-07-13:
Thanks. 

FYI

Numerous OpAmp models that I tried years ago that failed under Qucs or QucStudio, work in Qucs-S/ngspice. This is a huge improvement.

ra3xdh, 2022-07-13:
Both Qucs and QucsStudio have simulation kernel designed from scratch. It has a well known problem with time-domain simulation https://github.com/Qucs/qucsator/issues/19 and requires a conversion from SPICE syntax. Ngspice has very good compatibility with industrial models if some specific LTSpice/PSpice extensions are not used.

tomhajjar, 2022-07-13:
The main problems I had for OpAmps was with Qucsconv. The version in Qucs is horrible and fails more than it works. The one in QucsStudio generally works but has problems with complex models. Both programs are garbage for Transient simulations. 

Many times, I had to convert a OpAmp model in QucsStudio and then edit the file to work in Qucs. The newest QucsStudio has changed the file format so much it is a waste of time.

closed 2022-07-18 by ra3xdh