ra3xdh/qucs_s #118: [question] How to add transistors that aren't in the list?

yurivict, 2022-07-14:
I tried to add the ```2N5484``` transistor. It is quite common, but it doesn't appear in the list. How to add such transistor?

ra3xdh, 2022-07-15:
You need to use `.MODEL` component. It could be found at the *Spice specific sections* group. This component could contain the SPICE modelcard.
[image]

yurivict, 2022-07-16:
This way there would be a lot of .MODEL elements in the schematics.

I think Qucs-S should contain the database of transistors like below, and should just allow the user to select it by the model:
[image]

ra3xdh, 2022-07-16:
> This way there would be a lot of .MODEL elements in the schematic.

This is done intentionally and allows to embed the models in schematic and to distribute schematic alone without the `*.lib` file.

> I think Qucs-S should contain the database of transistors like below, and should just allow the user to select it by the model:

If you don't like using models, You can use the `.INCLUDE` pseudo device instead. It contains reference to the model library.
[image]

yurivict, 2022-07-16:
You can include all models in the schematic without showing them in the window. This overcrowds the window.

ra3xdh, 2022-08-13:
I have added a commit https://github.com/ra3xdh/qucs_s/commit/1ac7450f3ef8625d823fab48a814cf6da2c318a5 resolving this issue. Now all model lines except the first line containing model name are hidden when the model is dragged from the library.
[image]

closed 2022-08-13 by ra3xdh