ra3xdh/qucs_s #119: Location of log file

tomhajjar, 2022-07-16:
Where is Qucs-S storing the ngspice error log file? Some files cause ngspice to crash and the on-screen messages disappear before they can be read.

ra3xdh, 2022-07-16:
The log file location is `$HOME/.qucs/spice4qucs/spice4qucs.log` Later it could be accessed by pressing F5 on keyboard or *Simulation->Show last messages* The netlist is `$HOME/.qucs/spice4qucs/spice4qucs.cir` (accessed by pressing F6).

But the Windows version has some unique features. The default Windows Ngspice build doesn't write the text log. It only shows it in the windows that is closed after the simulation is finished. You need to use `ngspice_con.exe` instead of `ngspice.exe` to see the text logs.
[image]

tomhajjar, 2022-07-16:
Shouldn't the "default" location be what is set in the "Qucs Home:" path? In my case I use .qucs - S not .qucs. I try to keep the two programs segregated from each other.  

[image]

ra3xdh, 2022-07-17:
Yes, I meant the "Qucs Home" path. Inside this directory is the default location for logs and netlist.

tomhajjar, 2022-07-17:
Something is wrong then. The only spice4qucs directory is in my ".qucs" directory and the files generated by Qucs-S are there
[image]
.

ra3xdh, 2022-07-17:
Yes, I wrote a wrong location of the log file. It should be saved in Qucs home directory `log.txt`. The default is `$HOME/.qucs/log.txt` The correct log could be viewed by pressing F5 key. You can change the directory where the netlist and SPICE supplementary files are stored using the *Simulation->Select default simulator* dialog.
[image]

ra3xdh, 2022-07-30:
I have changed the default Ngspice executable name for Windows from `ngspice.exe` to `ngspice_con.exe` to avoid such issues with the log file in the future.

closed 2022-07-30 by ra3xdh