ra3xdh/qucs_s #120: 555 timer models working in ngspice but don't work using Qucs-S

tomhajjar, 2022-07-19:
As an exercise I ran some simulations using 4 different 555 timer models. I used the DuSpice GUI to confirm all worked with ngspice. Only 1 model (UA555.lib) worked when run via Qucs-S. 

I suspect some of the issues are due to my Qucs-S netlists lacking the added "Steps" that the authors of the ngspice netlists used to get the models to work.

I only mention this because I would think new users might use a 555 timer and become frustrated.

 [555 Timer_examples_prj.zip](https://github.com/ra3xdh/qucs_s/files/9141612/555.Timer_examples_prj.zip)
[555 Timer ngspice netlists.zip](https://github.com/ra3xdh/qucs_s/files/9141615/555.Timer.ngspice.netlists.zip)

ra3xdh, 2022-11-07:
A library and/or example containing 555 timer model should be added to Qucs-S. This will help to new users.

tomhajjar, 2022-11-07:
It's good this has come up again. I was unable to get a number of the 555 models attached above to work in Qucs-S. They came from the ngspice examples folder. Some issues are caused by not exactly duplicating the test conditions in the ngspice netlists.

At a minimum bipolar and CMOS 555 timer examples should be made to work for Astable, Monostable and PWM configurations.

tomhajjar, 2022-11-17:
I took another look at the 555 timer folders I sent above. I only included four 555 models that I got to work under ngspice. I cannot get TLC555ng.lib to work under Qucs-S but does work under ngspice. I included all the ngspice files I ran using DuSpice.

[555 Timer_examples_prj.zip](https://github.com/ra3xdh/qucs_s/files/10035907/555.Timer_examples_prj.zip)

ra3xdh, 2022-11-18:
Hello Tom, thanks for providing examples. I will try to turn your 555 subcircuits into the library for distribution with Qucs-S.

tomhajjar, 2022-11-18:
Some background:

The 555 model, TLC555ng.lib is a modification by Holger of TLC555.lib supplied with ngspice. I don't understand the reason it was done since both TLC555ng.lib and TLC555.lib work with the test file 555-timer-2.cir also supplied with ngspice. I use DuSpice for verification. Neither TLC555ng.lib or TLC555.lib work using my schematics in Qucs-S. 

https://sourceforge.net/p/ngspice/discussion/ngspice-tips/thread/d2048be2/
https://github.com/imr/ngspice/blob/master/examples/p-to-n-examples/555-timer-2.cir

555-timer-2.cir uses two different 555 models, TLC555.lib and UA555.lib, CMOS and bipolar.

TLCng_UA555_astab.cir is my test file that removed the UA555 model from 555-timer-2.cir

[image]

[555 Timers.zip](https://github.com/ra3xdh/qucs_s/files/10044008/555.Timers.zip)

tomhajjar, 2022-11-18:
For some reason I am unable to edit the above comment. A lot of text is missing. I updated the zip file. 
[555 Timers.zip](https://github.com/ra3xdh/qucs_s/files/10044116/555.Timers.zip)

ra3xdh, 2022-11-19:
> I cannot get TLC555ng.lib to work under Qucs-S but does work under ngspice.

The oscillator circuit started to work after I replaced the `SpiceLibComp` device by `Spice netlist` device. `SpiceLibComp` was designed for directly placing on schematic and may produce nested subcircuits when wrapped inside subcircuit. Look at the attached screenshots. I will provide a 555 time library based on your examples and include it in the next release. 

[image]
[image]

tomhajjar, 2022-11-20:
I got another 555 model to work in Qucs-S via ngspice. The author Clyde claims the model works very well. The files originally used a format for commenting that is not supported by ngspice anymore. Clyde pointed out the issue and I made changes. 

https://sourceforge.net/p/ngspice/discussion/ngspice-tips/thread/f535c0f1b9/

[555 Timers.zip](https://github.com/ra3xdh/qucs_s/files/10048301/555.Timers.zip)

tomhajjar, 2022-11-20:
I got another 555 model to work in Qucs-S via ngspice. The author Clyde claims the model works very well and my testing shows Reset works like you would expect unlike most of the other models I sent.

The files needed changes to work with newer rev ngspice. Clyde pointed out the issue and I made changes. 

https://sourceforge.net/p/ngspice/discussion/ngspice-tips/thread/f535c0f1b9/

[555 behavioral model and files.zip](https://github.com/ra3xdh/qucs_s/files/10048319/555.behavioral.model.and.files.zip)

ra3xdh, 2022-11-25:
I think the library containing two 555 models will be sufficient. I am planning to take two implementations for this library:

* TLC555ng model from Holger
* 555 XSPICE model from Clyde

tomhajjar, 2022-11-26:
The 555 XSpice model by Clyde is being modified to work over a wider voltage range. Presently it works for only 5 volts. Not sure when he will finish it.

ra3xdh, 2023-01-16:
Yes, I tried this 555 model. It works fine with Ngspice+Qucs-S.

ra3xdh, 2023-02-02:
I have added a library containing 555 timer model and oscillator example. This will available since v1.0.1 release.

tomhajjar, 2023-02-02:
Both CMOS and Bipolar models?

ra3xdh, 2023-02-02:
A new library contains only CMOS model provided by Holger Vogt. I didn't add the bipolar model yet.

ra3xdh, 2023-02-02:
I have just updated the library and added the bipolar model from KiCAD forum.

ra3xdh, 2023-02-04:
A new v1.0.1 package (released today) contains 555 timer library. Closing this.

closed 2023-02-04 by ra3xdh

tomhajjar, 2023-03-29:
Clyde over at ngspice has improved his 555 timer behavioral model to work over a wide range of voltages. Previously it was locked to 5 volts. He breadboarded one configuration and confirmed the model worked.

https://sourceforge.net/p/ngspice/discussion/ngspice-tips/thread/f535c0f1b9/

[555behav_Oscillator.zip](https://github.com/ra3xdh/qucs_s/files/11103751/555behav_Oscillator.zip)