ra3xdh/qucs_s #122: Add potentiometer component

tomhajjar, 2022-07-22:
Is a potentiometer in a future release? Since log or alternate taper pots are pretty much gone today a simple liner one would suffice. The ability to alter resistance via a variable would be a plus.

ra3xdh, 2022-07-22:
No, I don't planning to add potentiometer device. Qucs-S has no interactive simulation mode so the potentiometer will be not so useful. For Ngspice it's need to emulate potentiometer using the connection of two resistors and define its resistances using the variable. It will be not easy to use this combined device in parameter sweep, because the parameter sweep is implemented using the `alter` command which cannot directly sweep variables. But I am planning to add a Tuner feature that can resolve some tasks for potentiometer. There was an unfinished tuner PR for Qucs https://github.com/Qucs/qucs/pull/471 it's possible to backport it into the Qucs-S.

tomhajjar, 2022-07-30:
I did some experimenting trying to get a potentiometer to work. Having issues passing multiple "W" values via Parameter Sweep. Not sure how to use .PARAM to do it.
 
[Potentiometer_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9224727/Potentiometer_Test_prj.zip)

ra3xdh, 2022-07-30:
The `.PARAM` section has effect only before the simulation. It is not passed to `.control .. .endc` section of the netlist, so it will not help you. The problem with potentiometer implementation comes from this fact that Ngspice has no separte parameter sweep analysis type (like `.SW` for Xyce) And Qucs-S needs to emulate the parameter sweep using auto-generated Nutmeg script. See https://github.com/imr/ngspice/blob/master/examples/various/param_sweep.cir for example. The `alter` command accepts only a single device and doesn't accept variables. It is fundamental limitation of the Ngspice. Probably something will be changed in the future, because Ngspice team is currently working on `.SW` simulation type. The similar problem exists with sweep and vector parameters like SIN source #53 

So you can sweep only device but not variable. Your schematic will not work and you need only to write a custom nutmeg script (*Simulation->Nutmeg script* device). Something like this. There is no other way to get the variable sweep with Ngspice.

~~~
let start_w = 0.1
let stop_w = 0.9
let delta_w = 0.1 
let w_act = start_w
let rpot = 1k
* loop
while w_act le stop_w
  let  r1_act = (1-w)*rpot 
  let r2_act = w*rpot
  alter r1 r1_act
  alter r2 r2_act
  write dc-sweep.out v(out)
  set appendwrite
  let w_act = w_act + delta_w
end
~~~

tomhajjar, 2022-07-30:
I'm missing something. Tried multiple changes but no Bueno

[Potentiometer_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9226858/Potentiometer_Test_prj.zip)
.

ra3xdh, 2022-07-30:
There was a typo in the script. This schematic works as expected.
[image]
[image]

tomhajjar, 2022-08-01:
Hate to beat a dead horse....

Something is either either wrong with my Nutmeg script or I am doing something wrong. I cannot reproduce your plot even though the console output shows "Vout" is correct, 0.1, 0.2...0.9. Netlist doesn't work running under the DuSpice GUI either

1. Is file dc-sweep.txt a temporary file? I can find Potentiometer_Nutmeg_custom.txt but not dc-sweep.txt. 
2. Why is w_act saved twice and r1_act not saved at all in the netlist? -> Potentiometer_Nutmeg_custom.txt V(Vout) VPr1#branch w_act r2_act w_act

[Potentiometer_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9234511/Potentiometer_Test_prj.zip)

ra3xdh, 2022-08-01:
The node name `Vout` doesn't match the node name in the script `out`. You need either to change node name on schematic or in script. This line should be:
~~~
write dc-sweep.txt v(vout)
~~~

> Why is w_act saved twice and r1_act not saved at all in the netlist?

Automatic variables recognition may fail sometimes. Just append the missing variables to the semicolon separated list manually.

ra3xdh, 2022-10-04:
Assume this issue has been resolved. Closing ticket.

ra3xdh, 2023-11-28:
It makes sense to add potentiometer after implementation of the tuner feature and adding `alterparam` support. Reopening this. The potentiometer device is planned for the next release.

ra3xdh, 2023-12-01:
I have added SPICE entry for the potentiometer. This device will be available since the next release.

ra3xdh, 2023-12-01:
[image]

closed 2023-12-01 by ra3xdh

BURN-MICROSUCK, 2023-12-01:
Yesssss THANK YOU SO MUCH kind sir!!! Any estimate when will the next update be available? ...and/or can i clone/checkout the source code with potentionmeter from git right now?

ra3xdh, 2023-12-01:
I am planning to make an update in January or February. Yes, you may compile the `current` branch for git. It contains the new device.

BURN-MICROSUCK, 2023-12-06:
Hello, i just compiled current branch, it does indeed have the potentiometer device but it's not possible to wire it, i think because the pins don't "align" to the grid. I was able to get the wires to cross once and it stayed connected, but this is unusable. How can i fix this?

ra3xdh, 2023-12-06:
> I'm presuming the same would go for inductors in series and parallel, making it impossible to simulate RC / RCL circuits

This simulation is correct. The ideal opamp has zero output resistance. The time constant of the output stage RC=0. The output signal doesn't depend on the capacitance. Try to replace the opamp with the model from the library. But it also may have very low output resistance. For example for LM358 is set to RO=50 Ohm in the model.  And RC gives 2.35us, which gives cutoff frequency 425kHz.

ra3xdh, 2023-12-12:
@BURN-MICROSUCK I have just added a fix for device rotation. See https://github.com/ra3xdh/qucs_s/issues/92#issuecomment-1852227360 You may compile and test the `current` branch.

tomhajjar, 2023-12-25:
I'm not getting the potentiometer working. I used the standard method from Qucs. 

[Tone_Controls_prj.zip](https://github.com/ra3xdh/qucs_s/files/13766909/Tone_Controls_prj.zip)

ra3xdh, 2023-12-25:
> I'm not getting the potentiometer working. I used the standard method from Qucs

Define the variable with .PARAM and everything works as expected. Also fix the R4 value. The automatic converter assumes that the value without unit (Ohm) uses SPICE notation and parses it as 1mOhm.
[image]

tomhajjar, 2024-02-16:
I converted a Qucs tone control project in December using the new potentiometer model. I don't remember uploading it.

[image]

[image]

[image]

[Tone_Controls_prj.zip](https://github.com/ra3xdh/qucs_s/files/14303906/Tone_Controls_prj.zip)