ra3xdh/qucs_s #126: Legacy support for Qucs Equations

tomhajjar, 2022-08-04:
I have a lots of older files with the "old" Qucs Equation that work using ngspice as the simulation engine. Although the preferred method is to use Nutmeg, is there a reason I cannot continue to use the old method in new schematics? I have to cut and paste Equation blocks from older schematics since it was removed. 

I realize there are instances where the Qucs Equations won't work but so far it hasn't been a big issue.

ra3xdh, 2022-08-04:
Yes, I have switched Qucs-S to the Nutmeg equations since 0.0.24 version. The main reason is to prevent user from insertion of SPICE-incompatible functions like `time2freq()` or `stoz()` and force to use SPICE notation for nodes (like `v(node`), because Qucs notation (like `node.v` `node.Vt`) combined with SPICE-incompatible functions may lead to cryptic errors that are hard to debug. The usage of non-SPICE syntax when using SPICE kernel may be confusing. But the Qucs equation still operate for backward compatibility.  

I removed the Qucs equation from the main panel, but forgot to add it somewhere to the components panel. I will introduce a special group *Equations* in the next release and place here all equation-like devices (`.PARAM`, Qucs eqation, Nutmeg eqation).

ra3xdh, 2022-08-06:
I have created a special device group for equations in the left panel. The Qucs legacy equation is located here. This fix will be available in the next release. 
[image]

closed 2022-08-06 by ra3xdh