ra3xdh/qucs_s #127: Window select of more than 1 item doesn't work with mouse right click

tomhajjar, 2022-08-04:
If I want to cut or copy more than one item, I must use the Edit menu. If I select a group of items, as soon as I try to right click the mouse, all but one item is selected. I must select the items and drop the Edit menu to cut or copy the entire group.

ra3xdh, 2022-08-05:
The old Qt4 application has the same behavior, so this issue is not related to Qt5 porting. The right click on the component group doesn't work, but the keyboard shortcut `Ctrl+C/Ctrl+V` works as expected and allows to make copy/paste operations without need to access the main menu. I would advice you to use the keyboard shortcut for purpose too.

ra3xdh, 2023-02-19:
The implementation of this improvement is not possible without a redesign of `mouseactions.cpp`. I don't expect this to be done in the near future. Using keyboard shortcuts will resolve this issue for now. Closing this.

closed 2023-02-19 by ra3xdh