ra3xdh/qucs_s #128: Spice FET symbols

tomhajjar, 2022-08-05:
Previously I asked about having the correct Spice symbols for N/P enhancement and depletion power MOSFETs with 3 pins, (D-G-S) to match up with Spice models provided by manufacturers. The symbols exist in the Qucs legacy library but don't lend themselves to using Include...libraries. 

Another issue is identifying the Source lead of the Spice symbols for N/P MOS and JFET. The old Qucs method added a slash on the Source pin. Am I to assume the text near the pin always means Source?

[image]

ra3xdh, 2022-08-06:
> Another issue is identifying the Source lead of the Spice symbols for N/P MOS and JFET

Yes, the source pin for the SPICE JFET is the pin with the "NJF/PJF" text. I can correct the symbol and add slash on source pin to unify the symbol view.

> The symbols exist in the Qucs legacy library but don't lend themselves to using Include...libraries

The symbols from libraries (*.lib) files is not related to the symbols on the left panel and cannot be directly used as the system symbols. It's need to add new devices to implement the 3-pin MOSFET. Another problem is that such MOSFETS are not represented as MOSFET device (M letter) but as subcircuit (X device). Also some PDK for IC design like Google Skywater may define the 4-pin MOSFETS as the subcircuit device (X),  It's need to define the following new symbols:

* NMOS/PMOS 3-pin letter X
* NMOS/PMOS 4-pin letter X

ra3xdh, 2022-08-13:
I have added slash on source pin of the MOSFET symbols.
[image]

ra3xdh, 2022-08-13:
I have added all required MOSFETs by the commit https://github.com/ra3xdh/qucs_s/commit/aa8996e4971ba9daaa544e0ee63f7c40c425177c Now all combinations of 3/4pin M/X devices should be available. The attached screenshot illustrates the usage of the 3-pin MOSFET symbol for VDMOS device simulation according Ngspice manual.
[image]
[image]

closed 2022-08-13 by ra3xdh

tomhajjar, 2022-08-15:
So the added devices include both depletion and enhancement mode devices? 

Enhancement mode 3-terminal n/p power Mosfets most popular. 

Depletion mode 3-terminal n-channel power Mosfets have become common.

ra3xdh, 2022-08-16:
I am not inclined to add a separate symbols for enhancement/depletion devices. Ngspice shares the same `.MODEL` syntax for both MOSFET types and some proprietary software also use the common symbol for all MOSFETs. If we add a separate symbol for depletion and enhancement devices there will be 16 symbols. This will overcrowd the components panel and may be misleading for users.