ra3xdh/qucs_s #130: Getting ngspice examples to work via Qucs-S and proper configuration of MOS inverter

tomhajjar, 2022-08-06:
ngspice BSIM examples that work using DuSpice /ngspice are not working using Qucs-S. Two issues came up:

1. BSIM4_prj - BSIM4 files work using ngspice but under Qucs-S produce "Alberto says:  YOU TURKEY!  The limiting function received NaN." and generate no data. There are a number of posts in the ngspice forum on this problem but the files work under the latest ngspice release. https://sourceforge.net/p/ngspice/discussion/133842/thread/fa1aa20c/

2. I cannot get any n/p MOS inverter to work in Qucs-S if the top p-device's sub pin is connected to gnd. The p-device won't pull up to the + rail when the input is 0. It only works connecting sub and source together. In ngspice, the inverter works with the p-device sub pin connected to gnd.

[image]

[image]
[image]
[BSIM examples.zip](https://github.com/ra3xdh/qucs_s/files/9275125/BSIM.examples.zip)

ra3xdh, 2022-08-06:
Hello Tom, thank you for testing the BSIM models operation with Qucs-S.

> ngspice BSIM examples that work using DuSpice /ngspice are not working using Qucs-S

Could you point what exactly netlist is working with console Ngspice (DuSpice) and doesn't work with Qucs-S? The archive contains seceral projects. I have tested your project and get the same result from console and Qucs-S. 

> I cannot get any n/p MOS inverter to work in Qucs-S if the top p-device's sub pin is connected to gnd. 

The connection of the bulk node of the PMOS to GND is wrong. The circuit may converge in Ngspice, but the current will always flow through the open drain-bulk parasitic diode that is not acceptable.

ra3xdh, 2022-08-06:
I have found a problem with your BSIMSOI inverter. The device geometries (W and L) are missing for MOS transistors. I suspect that Ngspice will fail if no valid geometries are defined for the device. The inverter example starts to operate after the following correction.  
[image]
[image]
[image]

ra3xdh, 2022-08-06:
I have noticed that the MOSFET geometries (W and L) are also missing for BSIM3 and BSIM6 examples too. This also may cause convergence problem.

tomhajjar, 2022-08-06:
"Could you point what exactly netlist is working with console Ngspice (DuSpice) and doesn't work with Qucs-S?"

The BSIM4 libraries provided by the ngspice team work for all the netlists provided in the attached file. ngspice doesn't seem to care that the device geometries (W and L) are missing. I use the DuSpice GUI to call ngspice. Not sure why Qucs-S would care what device parameters are missing. Isn't that the job of ngspice?

[ngspice BSIM4 test files.zip](https://github.com/ra3xdh/qucs_s/files/9275818/ngspice.BSIM4.test.files.zip)

tomhajjar, 2022-08-06:
"The connection of the bulk node of the PMOS to GND is wrong. The circuit may converge in Ngspice, but the current will always flow through the open drain-bulk parasitic diode that is not acceptable."

I cannot find examples where the pMOS sub/well/bulk pin in a simple inverter should be connected to VDD/VCC instead of Ground. Every example online seems to connect the pin to ground. The examples supplied with ngspice connect the pin to gnd. Very confusing...

[image]
[image]
/

tomhajjar, 2022-08-06:
Do you have libs for BSIM3, 4, 6(BULK) that I can use that work with ngspice and Qucs-S?

tvrusso, 2022-08-07:
Just a quick comment:  The error message "Alberto says: YOU TURKEY! The limiting function received NaN." is coming from ngspice itself, not qucs-s.  That error message comes from the BSIM4 SOI code:  https://sourceforge.net/p/ngspice/ngspice/ci/master/tree/src/spicelib/devices/bsimsoi/b4soild.c around line 86.  

As Vadim says, the problem is that you are not providing L and W values on the instance line.  That "LLONG" and "WWIDE" appear in the model card is not contradicting that --- L and W are not on the instance line, and that's a problem.

Vadim's [fix ](https://github.com/ra3xdh/qucs_s/issues/130#issuecomment-1207257702) above puts L and W on the instance line by attaching them to the schematic symbol.  These are not model parameters, and would not be part of the model cards that the BSIM group provides.  They are instance parameters and are attached to the specific MOSFET device "M" line.

In every one of the files you attached in ngspice.BSIM4.test.files.zip, W and L are specified on the M lines:

```
Inv_chain.cir:MN0 out0 in VS VS VS N1 W=5u L=0.18u
Inv_chain.cir:MP0 out0 in VD VS VD P1 W=10u L=0.18u
Inv_chain.cir:MN1 out1 Out0 VS VS VS N1 W=5u L=0.18u
Inv_chain.cir:MP1 out1 Out0 VD VS VD P1 W=10u L=0.18u
Inv_chain.cir:MN2 out2 Out1 VS VS VS N1 W=5u L=0.18u
Inv_chain.cir:MP2 out2 Out1 VD VS VD P1 W=10u L=0.18u
Inv_chain.cir:MN3 out3 Out2 VS VS VS N1 W=5u L=0.18u
Inv_chain.cir:MP3 out3 Out2 VD VS VD P1 W=10u L=0.18u
Inv_chain.cir:MN4 out4 Out3 VS VS VS N1 W=5u L=0.18u
Inv_chain.cir:MP4 out4 Out3 VD VS VD P1 W=10u L=0.18u
inv_dc.cir:MN0 Out Gate VS VS VS N1 W=10u L=0.18u
inv_dc.cir:MP0 Out Gate VD VS VD P1 W=20u L=0.18u
inv_ring_51.cir:mn1  out in  ss  sub  n1  w=4u  l=0.15u  AS=6p AD=6p PS=7u PD=7u pdbcp=0u
inv_ring_51.cir:mp1  out in  dd  sub  p1  w=10u l=0.15u  AS=15p AD=15p PS=13u PD=13u pdbcp=0u
inv_tr.cir:MN0 Out Gate VS VS N1 W=10u L=0.18u Pd=11u Ps=11u
inv_tr.cir:MP0 Out Gate VD VS P1 W=20u L=0.18u Pd=11u Ps=11u
```

tomhajjar, 2022-08-07:
Thank you. I was blind to the obvious. 

Qucs removed all the BSIM examples and models years ago so I didn't get much exposure to it.

ra3xdh, 2022-08-07:
> I cannot find examples where the pMOS sub/well/bulk pin in a simple inverter should be connected to VDD/VCC instead of Ground. Every example online seems to connect the pin to ground

The PMOS bulk should be connected to VDD. Otherwise the parasitic diode will always be opened. For example look at this manual https://courseware.ee.calpoly.edu/~dbraun/courses/ee307/F02/02_Shelley/Section2_BasilShelley.htm Also you may refer to book from cmosedu.com for more examples of IC desing: http://cmosedu.com/cmos1/book.htm

> Do you have recommended libs for BSIM3, 4, 6(BULK) that work with ngspice and Qucs-S?

You can use the libraries from CMOSedu to test the BSIM4 model operation with Qucs-S http://cmosedu.com/cmos1/cmosedu_models.txt

tvrusso, 2022-08-07:
If you look at the netlists in your zip file, @tomhajjar, you will see that all the inverter netlists you have already seen work in ngspice directly have the bulk nodes tied as Vadim suggests --- the PMOS bulks are all tied to VD, and the NMOS bulk VS.

tomhajjar, 2022-08-09:
I went down the rabbit hole trying to reproduce data from this old paper and from folders I found in old releases of Qucs. The Qucs team removed all the BSIM devices/models from future releases.

[BSIM Projects.zip](https://github.com/ra3xdh/qucs_s/files/9286040/BSIM.Projects.zip)

[Mike_Brinson_MOS-AK_Venice_2014.pdf](https://github.com/ra3xdh/qucs_s/files/9286026/Mike_Brinson_MOS-AK_Venice_2014.pdf)

ra3xdh, 2022-08-09:
> ngspice has BSIM6/BULK libs/examples/cards but release 37 won't work with them

The BSIM >=4 may be disabled for default Ngspice build. You may need to need to rebuild Ngspice to enable its support. Need more investigation on this.

ra3xdh, 2022-08-09:
> I went down the rabbit hole trying to reproduce data from the attached old paper and from examples I found in old Qucs source.

The CMOS inverter circuit on the slide 13 of the presentation has a wrong connection of the PMOS bulk. This example may converge and seem to work correctly, but its operation is total wrong.

tomhajjar, 2022-08-09:
Here is the error message when I try and run the ngspice supplied BSIM6/BULK examples. I get similar messages for other examples using different device models. Not sure why they would disable so many devices from working. I don't know if having "set ngbehavior=ltpsa" as my default has any effect on the models one can use.

I cannot get any of the old Qucs BSIM6 examples to work properly using the modelcards supplied with them. The PMOS bulk connection isn't the issue. The devices exhibit low transconductance compared to the plots supplied with the examples. Maybe I'm not passing a required parameter.
  
[image]
[image]
[image]

ra3xdh, 2022-08-09:
I have made a bit more investigations. The warning is coming from Ngspice and says that the MOSFET model level 77 is not supported by this build. According the chapter 11.2 of the manual https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-12883 Ngspice supports only BSIM3 and BSIM4 models. The BSIM6 is not listed in the table. Furthermore the BSIM6 is missing in the Ngspice sources. It seems that Ngspice doesn't support BSIM6 model and BSIM6 example will not work with Ngspice. You need to install XYCE https://xyce.sandia.gov/ to make BSIM6 models work. According the reference guide Xyce supports BSIM level 77 models.

tomhajjar, 2022-08-09:
Thank you. I have had xyce installed for many years but rarely used it. By coincidence I updated to the latest version last week.

tomhajjar, 2022-08-09:
I redid the "BSIMBULK" test files to use xyce, 

The n/p inverter transient results look very strange if the input square wave has 1ns trise/fall. Not sure why but a SIN input was used by both the original Qucs authors and in the ngspice test files. Am I missing something about BSIM6/BULK?

[BSIM6_xyce_prj.zip](https://github.com/ra3xdh/qucs_s/files/9292753/BSIM6_xyce_prj.zip)
[image]
[image]

ra3xdh, 2022-08-09:
BSIM6 model is optimized for submicron technology and may show incorrect result for long channel (L=10u). If I set the device dimensions W=1u L=0.18u everything works as expected. 
[image]

tomhajjar, 2022-08-09:
Haaa burned again by test files. I used the L/W in the Qucs and ngspice test files.

tvrusso, 2022-08-09:
There are a number of BSIM6 test cases in the Xyce Regression suite's BSIM6 directory.  These are basically copies of stuff from the BSIM group, and you can find them [here](https://github.com/Xyce/Xyce_Regression/tree/master/Netlists/BSIM6).  They oddly use the sin source just as you saw in Qucs and ngspice, so they probably grabbed them from the BSIM group, too.

tomhajjar, 2022-08-10:
BSIM-BULK/6 lib from ngspice examples only works with xyce. The lib was the same in both packages. 

EKV2.6 lib from the ngspice examples wouldn't work for ngspice or xyce. EKV3 lib file from the xyce suite worked with xyce. 

[BSIM6-EKV3_xyce.zip](https://github.com/ra3xdh/qucs_s/files/9295374/BSIM6-EKV3_xyce.zip)

ra3xdh, 2022-08-10:
> EKV2.6 lib from the ngspice examples wouldn't work for ngspice or xyce

EKV2.6 model is not supported by default Ngspice build and will not work. It requires to rebuild Ngspice with `--enable-adms` option.

tvrusso, 2022-08-10:
The ekv 2.6 model in ngspice  is not the same as the "official" one that was only recently released into open source (https://github.com/ekv26/model) -- it is a reverse-engineered version constructed from published papers about EKV.  Xyce obtained the official sources years ago under an NDA and that is what is in our "non-free" binaries.  There are no model cards distributed by the authors of EKV2.6 that are usable to verify the implementations, and it would not surprise me that ngspice's example models don't work in Xyce.

The EKV3 model in Xyce is similarly provided from Technical University of Crete under an NDA, which is why it is in our "non-free" binaries but not the open source sourcecode in our repos.  The model cards in our test suite are taken directly from the EKV3 web site's examples.

tomhajjar, 2022-08-10:
I tested a few more devices from the Xyce Regression suite. BSIM-MG and PSP. 

BSIM-MG seems to work. The two netlists connect pch sub to +. I tried gnd and + and it didn't effect data. I guess the model doesn't care. 

PSP has odd Id/Vd curves. I tried multiple W/L from some papers but the Id/Vd only got more distorted. 

[MOS_Inverter_xyce_prj.zip](https://github.com/ra3xdh/qucs_s/files/9302935/MOS_Inverter_xyce_prj.zip)

ra3xdh, 2022-09-10:
Assume the questions regarding the BSIM4 support has been resolved. Closing this ticket. I have uploaded examples based on project provided by @tomhajjar to prevent such issues in the future.

closed 2022-09-10 by ra3xdh