ra3xdh/qucs_s #131: Subcircuit parameters support is broken?

denizzzka, 2022-08-09:
I am not sure this is my mistake or bug?

```
Ngspice started...
warning, can't find model 'rv' from line
    r1 _net0 _net1 rv
```
[image] [image]
[image]

ra3xdh, 2022-08-09:
I cannot reproduce your problem with Qucs-S+Ngspice-37. The circuit is simulated correctly on my machine. What Ngspice version are you using? Also try to wrap the variable in braces as shown. This will force Ngspice that this is variable but not model. Another solution is to use the blue resistors instead of red. For blue resistors Qucs-S wraps the value defined as variable in braces automatically. 
[image]

denizzzka, 2022-08-09:
ngspice-37 is used. Yes, circuit works with {} braces!

PS: Omg, here is three resistors types! :-)

closed 2022-08-09 by denizzzka