ra3xdh/qucs_s #132: Installation of Qucs 24 win64 versus Qucs 19S win32

tomhajjar, 2022-08-11:
qucs_s_win64_0_0_24.zip doesn't contain a setup file, only /bin and /share folders. I overwrote the /bin and /share folders into C:\Program Files (x86) directory originally done by Qucs 19S win32 install. There is no Qucs-S installation in C:\Program Files

Everything works, but is there any reason there isn't a win64 installation? 

In this case does Qucs-S v24 win64 run in 64 bit mode?

ra3xdh, 2022-08-11:
> qucs_s_win64_0_0_24.zip doesn't contain a setup file

Yes, I don't provide the installer anymore. The advantage of the portable package is that works everywhere without installation. The installer requires to be system administrator to install the software.  

> In this case does Qucs-S v24 win64 run in 64 bit mode?

Yes, I don't provide 32-bit Windows binaries since v0.0.23. There is only 64-bit Windows package available. The 32-bit PCs are almost extinct nowadays, so I don't see problem with 64-bit only package.

tomhajjar, 2022-08-11:
Understand. Much better anyway to get around issues writing to the programs folder.

Is there a reason you never included icons for desktop shortcuts? I always copied a new folder containing icons from older Qucs installations. 

[qucs icon.zip](https://github.com/ra3xdh/qucs_s/files/9311413/qucs.icon.zip)

ra3xdh, 2022-08-12:
Yes, I will include the provided icons in WIndows package in the next release.

ra3xdh, 2022-08-16:
I have prepared a 32-bit Windows portable package. You can download it here and use if necessary: https://github.com/ra3xdh/qucs_s/releases/download/0.0.24/qucs_s_win32_0_0_24.zip

closed 2022-08-27 by ra3xdh