ra3xdh/qucs_s #133: xyce examples don't work

tomhajjar, 2022-08-11:
I found a number of the supplied xyce Examples that don't work. Attached are few I fixed.

Does xyce have the unwrap phase function available? Ultimately I want to be able to generate Group Delay for S-parameters. 

[2N2222_Amplifiers_xyce_prj.zip](https://github.com/ra3xdh/qucs_s/files/9312097/2N2222_Amplifiers_xyce_prj.zip)

tomhajjar, 2022-08-11:
I found two schematics used R1=39K Ohm instead of 39k Ohm. 

Another issue caused xyce simulation to be wrong and Qucs-S to randomly crash.

R5=1M Ohm worked
R5=1MEG  worked
R5=1M bad news

tomhajjar, 2022-08-12:
[Temp2.zip](https://github.com/ra3xdh/qucs_s/files/9313198/Temp2.zip)

ra3xdh, 2022-08-12:
>  the supplied xyce Examples that don't work

Could you point exactly what doesn't work? The schematics from attached project are simulated without error except the issue with units (see my explanations below).

> Does xyce have the unwrap phase function available?

Xyce has very poor implementation of the postprocessor functions compared with Ngspice. I cannot find something like `cph()` in the Xyce manual. You may achieve complex measurement like the group delay only with export of the dataset and processing it with external application. 

> Another issue caused xyce simulation to be wrong and Qucs-S to randomly crash.

I have tried different combinations of the resitance but I was not able to reproduce application crash. The `1 M` value is treated as 1mOhm because the SPICE is registry independent. That's why you obtain short circuit on output and nearly zero output voltage. 

The Qucsator was registry dependent unlike SPICE and used different unit suffixes for units. The logic of units in Qucs-S is the following:

* If no unit is specified like (1k 10m) assume registry-independent SPICE notation. The 1M will 1m=1e-3 but not 1e6. The 1Meg will 1e6
* If unit (Ohm, V, F, etc) is specified assume Qucsator notation like 1kOhm = 1000 Ohm 1MOhm = 1000000 Ohm.

tomhajjar, 2022-08-12:
I sent you the xyce examples after they were corrected and cleaned up. The two "bad" schematics used R1=39K Ohm instead of 39k Ohm. Attached are the original file folders and the entire test suite. I forgot where I downloaded them from. 

Qucs crashed a coupe of times when R5=1M. I cannot say what additional actions are needed to trigger the crash.

It would be nice if the examples used in qucs-help.readthedocs.io where available.  

[Xyce_Examples.zip](https://github.com/ra3xdh/qucs_s/files/9316108/Xyce_Examples.zip)
[Xyce_qucs-s_Examples-master.zip](https://github.com/ra3xdh/qucs_s/files/9316118/Xyce_qucs-s_Examples-master.zip)

tomhajjar, 2022-08-12:
I contacted Tom Russo about the errors.

tvrusso, 2022-08-12:
Note that Xyce_qucs-s_Examples is not a product of the qucs-s team, it is a personal project that I created to show Xyce users how to use Qucs-s with it.  You shouldn't open issue requests in the qucs-s repo when you find something that doesn't work for you unless it's specifically a problem with qucs-s itself.  I cannot reproduce anything you've mentioned regarding random crashes on those examples.  The things you reported on my repo are not in fact errors.

You also appear to have a misconception about what should and shouldn't work.

Xyce, like SPICE before it, has a case-insensitive netlist parser.

1M and 1m are precisely the same thing in SPICE and Xyce, which ignores the case.  That is why "1M" gets you 1milli not 1meg.   It is why SPICE (and Xyce) has always had "MEG" as the suffix meaning 1,000,000.  This is in the manual and documented as a common mistake, because so many people expect "1M" and 1m" to mean something different.  It has been true since the first version of SPICE was written in the 1960s (and is true because the first version of SPICE was written on punch cards in the 1960s and all subsequent versions bear the scars).

1K and 1k are the same thing and there is no "typo" to fix.  Perhaps the case matters to qucs-s, but it doesn't matter to Xyce.  If qucs-s emits either one into a netlist, Xyce will simulate it the same.

Xyce does not provide a post-processing capability similar to nutmeg and has never pretended to.  Any post-processing capabilities we have are provided by the Xyce expression library and are accessed via print line expressions.  We are open to requests for specific features from users and will prioritize them based on available staff and how well they fit with the needs of Sandia National Laboratories and our funding sources.  We have never had a request for an unwrap phase function before.  Such feature requests would be appropriate to the Xyce repos, not qucs-s.

tomhajjar, 2022-08-12:
I was able to reproduce the plots Tom provided ONLY if I use manual Limits. Plots on my PC are blank using "auto". The auto ranging is not working using this file under xyce. 

Side note, using Vcc=1 V the BJT is starved for current...

[Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9329498/Test_prj.zip)

tvrusso, 2022-08-13:
I cannot speak to why the autoranging of charts doesn't work for you, as they work just fine for me.  This may or may not be an issue in the Windows version of qucs-s that you are using, but it isn't a bug in Xyce.

However you are correct, the vcc in the two common emitter amplifiers examples (which are intended as demos of how to use Xyce with qucs-s, not how to build common emitter amplifiers) was incorrectly copied into qucs-s using the default voltage value for the source instead of the intended 12V DC.  It has been corrected.

ra3xdh, 2022-08-13:
> Plots on my PC are blank using "auto". The auto ranging is not working using this file under xyce.

I also cannot reproduce this. The autoranging works as expected with attached example. I need to install Xyce on my Windows machine and see what happens.

tomhajjar, 2022-08-14:
The auto/man plotting issue caused my initial confusion confusion. I assumed K/k and 1V/12V were issues for my blank plots when they weren't.

[amp_xyce_plot.zip](https://github.com/ra3xdh/qucs
[image]
[image]
_s/files/9333404/amp_xyce_plot.zip)

tomhajjar, 2022-08-14:
I moved schematic over to Qucs 0.0.20. I found auto scale also fails for this schematic. It turns out if I change the frequency sweep star frequency from 0.01 to 0.1, the issue is gone. Did same on Qucs-S (xyce) and same occurred. 

If f min is 0.02 or greater, plot is OK on both Qucs and Qucs-S.

ra3xdh, 2022-08-14:
I can reproduce the issue with the auto-scaling now using `wine` on Linux platform. Only Windows version is affected. For Linux version everything works as expected (see attached screenshot). It's strange because the diagram implementation has no platform-dependent code. 

 > moved schematic over to Qucs 0.0.20. I found auto scale also fails for this schematic. 

No, nobody reported about about autoscaling issues before. It is the first such report.

[image]

ra3xdh, 2022-08-14:
The summary:
* Windows version shows empty plot on this schematic. Ngspice is affected too;
* Linux version works as expected;
[image]

ra3xdh, 2022-08-14:
The correct link for the archive with the test schematic https://github.com/ra3xdh/qucs_s/files/9333404/amp_xyce_plot.zip

ra3xdh, 2023-01-04:
Windows-64bit auto-scaling issue has been fixed by #201 Tanks @zergud for debugging. This fix will be available since v1.0.1 release. Closing this ticket.

closed 2023-01-04 by ra3xdh