ra3xdh/qucs_s #134: LED library doesn't work for ngspice or xyce

tomhajjar, 2022-08-16:
Back in 2018 Tom Russo noted the LED library doesn't work for xyce

I altered the library to function. I don't understand why any of the libraries uses Ohms considering it can cause issues. I also deleted all "units", Ohm, v, uA, pF... and converted 50 MOhm to 50e6, 1pF to 1e-12, 1uA to 1e-6...etc.

I also deleted all parameters that caused warnings in ngspice and xyce. All had values of "0" anyway. 

xyce is still complaining that TEMP and AREA are missing but they aren't. Is xyce case sensitive"

I renamed the original LED library and generated a new one to be copied into the library directory. 

[LED_prj.zip](https://github.com/ra3xdh/qucs_s/files/9350657/LED_prj.zip)

ra3xdh, 2022-08-16:
Thank you for fixing the LED library. I will replace it in the application distribution. This may be partial solution for #13 until I will provide a fix for qucs2spice subsystem. 

>  I don't understand why any of the libraries uses "Ohm" considering it can cause issues

The library was designed around 2007 when no Qucs-S existed. And the units are not a problem for Qucsator.

tvrusso, 2022-08-16:
FWIW, SPICE3F5 has always ignored units after the multiplier, and so does Xyce.  So "3Ohm" and "3" are the same thing.  "3pf" and "3p" are the same. etc.  The issue of #13 is that qucs-s is skipping a step when it is using the legacy libraries so that stuff like "50 MOhm" (which in Qucs means 50 mega ohm) gets emitted as "50" instead.

The LED libraries were legacy Qucs libraries.

I have already stated that Xyce is case insensitive just as is SPICE.

When Xyce says "no model parameter TEMP found for model" it does NOT mean that no parameter was found in the model card, it means that the model card CONTAINS a TEMP parameter, but the model card type does not allow one.

TEMP and AREA are not model parameters for the SPICE diode, they are instance parameters and must be on the instance line (that is, attached to the schematic part, not the .model part). 

There are SOME spice variants that allow you to specify all parameters on a .model card and they work just the same as if they were on the instance line.  SPICE3F5 and PSpice, which were what Xyce was patterned on, do NOT.  It is not OK in Xyce to stick instance parameters into the model card, they have to be on the instance line of the netlist.

Somehow these LED libraries are putting the TEMP and AREA parameters on the wrong netlist line.

The Xyce reference guide has tables for every single supported device stating which parameters are recognized as instance and which are model.  There are rare cases of devices with identically named instance and model variables, such as L and W --- in these cases when L and W are given in the model card and not on the instance line, they are copied into the instance variables and used.  If given on the instance line they are always used and the version in the model card is ignored.  In the cases where this is supported it is documented in the reference guide.

tomhajjar, 2022-08-16:
In this case, the LED models are OK approximations where Temp and Area are not needed anyway. 

If the LED is "special", one should use the actual Spice model of the device in question.

tvrusso, 2022-08-16:
Yeah.  AREA is a diode parameter that was invariably used to emulate multiple diodes in parallel rather than being a real parameter --- and as such is much better suited for the more modern use of a multiplicity factor (usually "M").  AREA is 1.0 by default, and it's being set to 1.0 in these models.

TEMP, on the other hand, could matter more.  If one has extracted parameters at a given temperature TNOM, then running it at any other temperature value means the parameters are all scaled internally by an appropriate factor (usually some power of TEMP/TNOM, but sometimes something more complicated).  If these models have TNOM equal to the temp set in the model card, and are then run in a simulator that is ignoring the TEMP setting, AND that temp setting is not the same as the default simulator temperature, then results will be ever so slightly different.

I do agree, though, if one really wants accurate predictive models of a real circuit, then one needs accurate SPICE models, and those are usually best obtained from the part manufacturer.  These LED models (once patched up so they actually work) are close enough for back-of-the-envelope simulation.  That's what I had wanted them for when I opened #13, and I wound up just coding something up myself for the quick one-off simulation I needed (probably just yanked the model parameters out of the Qucs library and stuck them in a model card myself).

tvrusso, 2022-08-16:
FWIW, I just looked at the ngspice source code for the diode, and it does not appear that *it* recognizes TEMP or AREA as a model parameter, either.  It is probably also ignoring them.

tomhajjar, 2022-08-16:
Hmmmm might need to check one component each semi library....

tomhajjar, 2022-08-16:
Testing a selection of diodes from the library with ngspice and xyce didn't generate any warnings. 

[Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9354459/Test_prj.zip)

ra3xdh, 2022-08-19:
I have replaced the LEDs library in the Qucs-S distribution. Now it should work for Ngspice/Xyce without simulator warnings.

closed 2022-08-19 by ra3xdh