ra3xdh/qucs_s #135: Add symbol preview for the library tree

ra3xdh, 2022-08-16:
Suggested by @tomhajjar  See https://github.com/ra3xdh/qucs_s/issues/92#issuecomment-1215515439 for context. Adding symbol and device info preview may improve the application usability.

tomhajjar, 2022-08-19:
Big improvement. I don't have to place a component and look at it's parameters, delete it and pick another....

ra3xdh, 2022-08-19:
The next step may be to remove the separate Qucs-lib tool, because the library tree on the left panel now has all features of this tool.

tomhajjar, 2022-08-19:
I've not found it useful. 

One thing "Qucs" lacks is the ability to easily make your own libraries or edit existing ones to add new devices. I've done it manually using a text editor but it's not fun. A big stumbling block was the Qucs format for it's files. Very different than Spice models and easy to screw up.

ra3xdh, 2022-08-19:
> One thing "Qucs" lacks is the ability to easily make your own libraries or edit existing ones to add new devices

Yes, I have requests from other users regarding the library editor as a separate tool. It is a more complex task and the most probably will not be implemented in the near future. 

> I've done it manually using a text editor but it's not fun.

Are you aware about the possibility to create libraries from the *Project->Create library* menu? There also exist `qucsconv` utility that was not ported to Qucs-S distributions and relates to Qucsator simulation kernel. It can convert SPICE library containing only `.MODEL` modelcards to Qucs XML library.

tomhajjar, 2022-08-19:
I am aware of it but does it allow editing or adding components?

Qucsconv in Qucs is abysmal. It fails more than 50% of the time. Even when you think it works, it sometimes has missing or corrupted devices. I always ended up using the Qucsconv in QucsStudio then edit the sch in a text editor to work in Qucs/Qucs-S. The most recent version of QucsStudio 4.2.2 has altered the sch file format so much it takes far too long to edit the files to be worthwhile.

ra3xdh, 2022-08-19:
> I am aware of it but does it allow editing or adding components?

Yes, currently there is no way to edit the library except using the text editor. The library editor tool would resolve this task.

tomhajjar, 2022-08-19:
So one generates a lib for each device then splices them together using a text editor?

tomhajjar, 2022-08-19:
I did a test of making a library. Got a weird error. R3 has no digital model. Why R3?

[image]
[Speaker_model_library.zip](https://github.com/ra3xdh/qucs_s/files/9384359/Speaker_model_library.zip)

ra3xdh, 2022-08-19:
> So one generates a "lib" for each device then splices them together using a text editor?

No, Qucs-S can combine the models from all subcircuits in project into one library. It's not need to combine it manually. 

> I did a test of making a library. Got a weird error. R3 has no digital model. Why R3?

This error could be ignored. The library generation procedure was inherited from Qucs that tries to generate digital model for all devices. The resulting `*.lib` file is correct.

tomhajjar, 2022-08-19:
Well Qucsconv is a massive failure. Converted 11 Op Amps and made a library, OpAmp.lib. 

Only the 2 NE5532s in the library worked. Pretty much mirrors my experience with Qucsconv. 

I used Op-Amp_Model_Test.sch to compare the "schematic" version of the Op Amps with versions in the library I created.

[image]

[Op Amp Spice Models_prj.zip](https://github.com/ra3xdh/qucs_s/files/9384713/Op.Amp.Spice.Models_prj.zip)
[OpAmp.zip](https://github.com/ra3xdh/qucs_s/files/9384715/OpAmp.zip)

ra3xdh, 2022-08-20:
> Well Qucsconv is a massive failure. Converted 11 Op Amps and made a library, OpAmp.lib

Qucsconv has several limitations. For me it works only for library containing modelcards `.MODEL` but always fails on subcircuits. I am inclined not to include qucsconv to Qucs-S distribution. My policy is to provide user a possibility to use unmodified SPICE device instead of using double conversion. SpiceLibComp and .MODEL devices were designed for this purpose https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#using-manufacturers-component-data-libraries

ra3xdh, 2022-08-20:
I have implemented the *Show model* button (see attached screenshot). I also modified the model info dialog to show Qucsator/Spice models and keep it up-to-date with Qucs-S.
[image]
[image]

tomhajjar, 2022-08-20:
Is there a way I can test the nightly release?

ra3xdh, 2022-08-20:
I can prepare a Windows build from the latest development state and share the portable package with you in the next week. I am planning also to add component search feature before preparing a beta package.

ra3xdh, 2022-08-20:
I have added a search button for the library tree on the left panel. It makes possible to search device over libraries (case insensitive).
[image]

closed 2022-10-30 by ra3xdh