ra3xdh/qucs_s #137: "BJT" examples in schematic\ngspice folder

tomhajjar, 2022-08-17:
I tested and cleaned up the "BJT" examples in schematic\ngspice folder. All simulated as shown in the online Help except "BJT-noise.sch". The noise level are orders of magnitude higher than shown in the Help. Either I made a typo or something is wrong...

https://qucs-help.readthedocs.io/en/spice4qucs/ASim.html

[BJT_examples_ngspice.zip](https://github.com/ra3xdh/qucs_s/files/9365892/BJT_examples_ngspice.zip)
[image]
[image]

ra3xdh, 2022-08-18:
Thanks for checking the BJT examples! I have also found that the model sweep syntax `T1.Bf` in `BJT_dc.sch` is broken. I have added a fix on the `current` branch. I will replace the examples in Qucs-S distribution to keep in up-to-date with the latest Ngspice. 

> The noise level are orders of magnitude higher than shown in the Help

I am obtaining the same values for `inoise` and `onoise`. The dataset produced by Ngspice is parsed correctly. The screenshot in the Help was taken from Ngspice verision 24 almost 7 years ago. Ngspice team has updated the noise simulation algorithm. And result may be different.

tomhajjar, 2022-08-18:
Is BJT_dc_ngspice.sch working properly using the sweep parameter @mod2N2222A[BF]?

[image]

tomhajjar, 2022-08-18:
I just noticed that I2 in BJT_dc.sch is backwards. Does this effect your comment? If I flip it, the plot looks like BJT_dc_ngspice.sch. 

[image]

Is BJT_dc_ngspice.sch working properly using the sweep parameter "@mod2N2222A[BF]"? 

In another example for xyce in web Help, the parameter is "mod2N2222A[BF]". If I remove the "@" ngspice crashes. 

[image]

tomhajjar, 2022-08-18:
Here is the xyce example. Can you explain why the online examples like below use a voltage source as the base drive?

[image]

ra3xdh, 2022-08-18:
> I just noticed that I2 in BJT_dc.sch is backwards. Does this effect your comment?

Yes, the current source polarity on this schematic was wrong. It outputs a proper IV-curve after the flipping the source.

> Is BJT_dc_ngspice.sch working properly using the sweep parameter "@mod2N2222A[BF]"?

Yes, this example is outdated. It requires the setting of the `SweepModel=True` parameter to work properly.

[image]

> Here is the xyce example. Can you explain why the online examples like below use a voltage source as the base drive?

The circuit with voltage source is technically correct and will converge. But you need to set the current source to obtain the proper IV-curves family. I cannot explain the intention of the author of this example to include the voltage source instead of current source.

ra3xdh, 2022-08-18:
> In another example for xyce in web Help, the parameter is "mod2N2222A[BF]". If I remove the "@" ngspice crashes.

The crash comes from Ngspice itself, not from Qucs-S GUI. If the `@` is removed the syntax of the `altermod` command become incorrect causing Ngspice error. Qucs-S cannot handle this automatically and prevent Ngspice crash.

tomhajjar, 2022-08-18:
I changed SweepModel to true for BJT_dc.sch and BJT_dc_ngspice.sch. Only BJT_dc_ngspice.sch seems to give the correct output. Is T1.Bf in BJT_dc.sch still the wrong syntax? 

[image]
[image]

ra3xdh, 2022-08-18:
> Is T1.Bf in BJT_dc.sch still the wrong syntax? What is the correct syntax?

The BJT_dc.sch schematic is affected by another bug. It's not possible to make it work with the `0.0.24` release. The fix will be available only in the next release. See my recent commit https://github.com/ra3xdh/qucs_s/commit/29a30058410ca09687097acb42c1795dae56081b

tomhajjar, 2022-08-18:
It's odd BJT_dc.sch generates a plot with a single sweep for Bf=20 and not the default 200.

[image]
[image]

ra3xdh, 2022-08-18:
> It's odd BJT_dc.sch generates a plot with a single sweep for Bf=20 and not the default 200.

Your calculation is wrong. The collector current should be Ic=Ib*Bf=5e-6*200=1e-3 A as shown on the graph.

tomhajjar, 2022-08-18:
Here is the latest and greatest. 

[BJT_examples_ngspice.zip](https://github.com/ra3xdh/qucs_s/files/9374297/BJT_examples_ngspice.zip)

tomhajjar, 2022-08-18:
I got BJT_dc.sch to work, Since the netlist name for T1 passed to ngspice is actually QMOD_T1, I just used @QMOD_T1[BF]. 

Is there a reason Qucs appends a prefix onto every library model designator? I assume the prefix is different for every device type in the libraries. Won't this complicate your fix? 

[image]

[BJT_param_sweep_ngspice.zip](https://github.com/ra3xdh/qucs_s/files/9374437/BJT_param_sweep_ngspice.zip)

ra3xdh, 2022-08-18:
> I got BJT_dc.sch to work, Since the netlist name for T1 passed to ngspice is actually QMOD_T1, I just used @QMOD_T1[BF]

Yes, this will work, but it is a hack and relies on internal Qucs-S netlister naming that is subject of change between the versions. And such example should not go to the documentation. The next release will provide a fix that will make the `Device.Param` syntax operational again.

> Is there a reason Qucs appends a prefix onto every library model designator?

The `@` prefix is required by Ngspice `altermod` command when swweping parameters inside the modelcard `.MODEL`. See the chapter 17.5.4 of the Ngspice manual for more details https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-24272 Everything comes from this fact that Ngspice has no `.SW` simulation type like Xyce and it's need to use hacks with `alter/altermod` and `while` loop to emulate the parameter sweep.

ra3xdh, 2022-08-26:
I have updated BJT simulation examples based on your project. Closing this.

closed 2022-08-26 by ra3xdh