ra3xdh/qucs_s #140: Making user libraries

tomhajjar, 2022-08-20:
I investigated further the issues I was having making an Op Amp user library. For the Op Amps I chose, if I make a library of a single Op Amp, it works in my circuit. If I try and make a library of more than one Op Amp, some don't work. When compare the files I can't see the reason for the issues. I'm using Qucscov from Qucs 0.0.20.

I made a simple test circuit. Ignore the crappy cross over distortion of the LM358.

[user_lib.zip](https://github.com/ra3xdh/qucs_s/files/9386099/user_lib.zip)
[Op Amp Library Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9386100/Op.Amp.Library.Test_prj.zip)

ra3xdh, 2022-08-20:
I have tried to create a library from your project and found no issues. Everything works as expected. After inspection of your library file I suspect that it was either renamed with file manager or edited with text editor. 

The components iside the library have the following name format `OpLib_component_name`. But the library name is `OpAmp.lib` Qucs-S netlister makes subciruit name for LibComp as following `LibName_CompName`. If the file name and subciruit name inside library are not matching you will get unknown subcircuit error. Please ensure that the library name and subciruit name are matching and the library will work. 

Also I advice to avoid spaces in subcircuit and component names when making library. It also may cause issues with Ngspice, because spaces are not allowed in subcircuit `.SUBCKT` name.

ra3xdh, 2022-10-19:
Assume the problem with creating a user library has been resolved. Closing this.

closed 2022-10-19 by ra3xdh