ra3xdh/qucs_s #141: Qucs-S analysis versus LTSpice

tomhajjar, 2022-08-22:
I "ported" a LTSpice file to Qucs-S to compare data. I ran into a number of issues. 

Qucsconv from Qucs 0.0.20 didn't work. The one from QucStudio did. 

When you cut and paste groups of components that only have nets attached, the netlist names get changed on the fly causing conflicts. I had to use keyboard "arrows" to move individual and component groups into approximate position before drawing wires to connect nets. Not really an issue for most users. 

It turns out only being able to rotate a resistor or capacitor only "left" makes for a lot of work. I vote for adding rotate "right"

Qucs has issues with a number of Spice models that work in LTSpice. Could not get LT1056 op amp to work with ngspice. For BJT, MOSFET parameter issues, I edited out the parameters ngspice didn't like. Complicated MOSFET models that worked with LTSpice didn't work in ngspice. I reverted to simple "VMOS" models.

The Op Amp I substituted doesn't work well in the circuit for a 10Khz input signal. Both LTSpice and Qucs-S show this. 

Bottom line is I was able to get pretty good agreement between LTSpice file and Qucs-S for a fairly complicated circuit. 

[Audio Amp Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/9397259/Audio.Amp.Test_prj.zip)
[Current Dumping Qucs-S.zip](https://github.com/ra3xdh/qucs_s/files/9397265/Current.Dumping.Qucs-S.zip)

holvo, 2022-08-26:
Complicated MOSFET models that worked with LTSpice didn't work in ngspice. I reverted to simple "VMOS" models.

Could you be more specific or give an example?

Could not get LT1056 op amp to work with ngspice.

Unfortunately Anallog Devices does not provide a model which is compatible to PSPICE, ngspice or other SPICEsetc. The LT1056 model can only be read by LTSPICE. Reason is the usage of

A1 2 1 0 0 0 0 0 0 OTA g=0 in=1.8f ink=100
...
A2 0 N003 0 0 0 0 X 0 OTA g=8.4u asym isource=3.28u isink=-9.84u Cout=205f en=15n enk=28 Vhigh=1e308 Vlow=-1e308

which are LTSPICE-specific. It might be possible to write a replacement for the A...OTA... model, someone has to take care.

tomhajjar, 2022-08-26:
Thanks for the feedback. It made me go back and check my work. I mistakenly used ".model" instead of spice file subcircuit for the MOSFETs. Attached is the correct implementation. The data mirrors using the .model MOSFET info. LTSpice users tend to use the internal simple .model MOSFETs. They are transitioning to a new .model with the "Ksubthres" parameter.

The only reason I used the LT1056 was it's used a lot on diyAudio.com for current dumping amps based on an old Elektor article. Unfortunately I haven't found a suitable replacement opamp that works either in LTSpice or Qucs-S. Most of the amps have a large output DC offset. and instability at 10KHz. Strangely some of the "ideal" opamps in the LTSpice UniversalOpamps2.sub library work. Qucs doesn't have a ideal opamp with Vcc/Vee. I will make one using the LTSpice library. 

[image]
[Amp_Current_Dumping_prj.zip](https://github.com/ra3xdh/qucs_s/files/9433524/Amp_Current_Dumping_prj.zip)

holvo, 2022-08-26:
You may have a look here: http://s-audio.systems/blog/spicemodels/

holvo, 2022-08-26:
Attached you will find a generic OpAmp model.
[uopamp_lvl2_test.zip](https://github.com/ra3xdh/qucs_s/files/9434084/uopamp_lvl2_test.zip)

tomhajjar, 2022-08-26:
I have them. They are in the original folder. None work like the LT1056 in this circuit. If you don't have high DC offset, you get instability with Fin =10KHz. Some don't work at all. They also "fail" in LTSpice. The best seems to be a "discrete" TL071 model I got off diyAudio but it also has issues at 10KHz.

As an aside, all of these amps pay homage to the Quad 405 current dumping amp from the 1970s. I built a 405 clone in 1976 in my dorm room. Etched the PCBs in the bathroom sink. It still works today. ;-)
  
[image]
[image]

[Op Amp Spice Models_prj.zip](https://github.com/ra3xdh/qucs_s/files/9434083/Op.Amp.Spice.Models_prj.zip)

tomhajjar, 2022-08-26:
I made a model of your universal amp. It work as a simple inverting amplifier but gives bizarre gain/phase response in the amplifier. It acts like a HPF. Can't see what I did wrong. 

[image]
[image]

[Amp_Current_Dumping_prj.zip](https://github.com/ra3xdh/qucs_s/files/9434351/Amp_Current_Dumping_prj.zip)

holvo, 2022-08-26:
That indeed does not look very reasonable. I will discuss it with the model devloper. Another simle model is attached.
[generic-hv.zip](https://github.com/ra3xdh/qucs_s/files/9435415/generic-hv.zip)
The results look promising:
[image]

I have made a setup to simulate a modern variant of the QUAD405 (Q17 designed by Tiberiu Vicol) with ngspice. You will find my setup (KiCad is used for schematic entry) at https://forum.kicad.info/t/simulation-examples-for-kicad-eeschema-ngspice/34443 . However I have never taken the chance to build such a thing.

tomhajjar, 2022-08-26:
Very few need an Opamp model that has accurate Vcc/Vee DC and AC currents. The original Quad 405 schematic doesn't. 

Decades ago people came up with "Op-amp boosters" and " Op-amp Bootstrapping". They need such models. Many of the current dumping amps on diyAudio are derivatives of the Elektor article. 

I read the Q17 postings when they came out and ran simulations. I had issues with Op Amp models then. I ended up using the TL071_Qlevel.sub model.

I built my Quad 405 clone using surplus junk I bought in Boston. I used the schematic from a Wireless World article on the 405. I wrote to Quad and they sent me an owners manual. I also ordered PCBs from England for the Douglas Self Advanced Pre Amp. My turntable was a kit from Popular electronics.  I had a Dremel and $8 Radio Shack VOM. The pre-amp and amp worked the first time I turned them on. The Quad clone survived many Dorm parties that would fry my Southwest Technical products amplifier. It still works today.

PCB layout and front panel renderings I did in class. instead of paying attention...
[image]

tomhajjar, 2022-08-26:
genericopa1.lib has two issues. 
Qucs-S cannot handle .params on  .SUBCKT line. Generates extra Pin called params.
.subckt genopa1 in+ in- vcc vee out params: POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg

[image]

When I alter the sub, the results are bad.
.subckt genopa1 in+ in- vcc vee out 
+ params: POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg

[image]

holvo, 2022-08-26:
You may do a change like this:
```
* generic OpAmp model
* gain, phase, offset, limits to power supply, input and output resistance
.subckt genopa1 in+ in- vcc vee out
.param POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg
Rin in+ in+ {RIN}
Voff in+ inoff dc {VOFFSET}
G10 0 int inoff in- 100u
R1 int 0 {GAIN/100u}
C1 int 0 {1/(6.28*(GAIN/100u)*POLE)}
Eout 2 0 int 0 1
Rout 2 out {ROUT}
Elow 3 0 vee 0 1
Ehigh 8 0 vcc 0 1
Dlow 3 int Dlimit
Dhigh int 8 Dlimit
.model Dlimit D N=0.01
.ends

```

tomhajjar, 2022-08-26:
Results the same as my modification of lib. 

Still cannot pass parameters changes from schematic to sub circuit. Wonder if that ever worked.

holvo, 2022-08-26:
Use the "code" button to mark ngspice input and output (to not be mangled by the text editor).

holvo, 2022-08-26:
This is in the docs:
https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#spice4qucs-subcircuits-with-parameters
 Is it still valid or no more working?

ra3xdh, 2022-08-27:
> This is in the docs:
This is valid for device combined from qucs devices (traditional subcircuit). Passing parameters to subcircuit represented by SPICE netlist works differently.

ra3xdh, 2022-11-28:
I have added a TODO list for this issue. I am planning to add TL071/TL072 devices to the `SpiceOpamp.lib` library.

ra3xdh, 2022-11-28:
I have just added TL071/TL072 model from http://s-audio.systems/blog/spicemodels/ to Qucs-S. Closing this ticket.

closed 2022-11-28 by ra3xdh