ra3xdh/qucs_s #143: New RF components

tomhajjar, 2022-08-25:
Any chance more RF "stuff" from Qucs 0.0.20 might end up in the next release? INDQ, CAPQ? S-parameters?

[image]

ra3xdh, 2022-08-25:
Yes, the INDQ and CAPQ devices are in my plan as the part of #94 I cannot say when exactly it will implemented in Qucs-S. The INDQ and CAPQ seem to easier as transmission lines. The s2p file device may have fundamental limitation that may make it incompatible with Ngspice. I am planning to start with INDQ and CAPQ and then continue with transmission lines.

tomhajjar, 2022-08-25:
Do you need the source...

[qucs-0.0.20-rc2.zip](https://github.com/ra3xdh/qucs_s/files/9426615/qucs-0.0.20-rc2.zip)

ra3xdh, 2022-08-25:
I can always take the Qt4 Qucs sources from Github https://github.com/Qucs/qucs The Qucs sources is a little help with implementation of the RF devices with Qucs-S+Ngspice. The S-parameter simulation works differently in Ngspice. The transmission line devices should be converted to LTRA device model https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-6082 The INDQ and CAPQ should be converted to subcircuits.

tomhajjar, 2022-08-25:
The L/C "Q" models don't include resonance but are much better than nothing. 

Manufacturers resonance data doesn't take into account how the component is used and mounted onto the PCB so can be misleading. I measure inductors myself at operating frequency. Measurements rarely equal manufacturer data.

rjordans, 2022-11-23:
selecting qucsator as the default simulator did allow me to use s2p files, but it's a bit annoying that I have to set this globally.  Would be nice if this could be set on a per-project basis

ra3xdh, 2022-11-23:
@rjordans The setting of the simulator per project/schematic is not possible due to some architecture limitations of the GUI. The GUI doesn't support the reload of the components palette on the fly which is needed after the simulator has been changed. So, only the global setting of the simulator is possible.

rjordans, 2022-11-23:
Ah, that's too bad then.  Thanks for the quick response!

tomhajjar, 2022-11-23:
@ra3xdh

Presently Qucsactor is not included with Qucs-S. Is that going to change? If so will Qucsactor be updated to the latest revision?

ra3xdh, 2022-11-23:
> Presently Qucsactor is not included with Qucs-S. Is that going to change?

No, Qucsator is maintained by another team https://github.com/qucs/qucsator and cannot be included in Qucs-S distribution. Qucsator team doesn't provide binary packages. But one can download and compile it manually if necessary.  Ngspice will remain the default simulator for Qucs-S, because Qucsator has numerous issues with TRAN simulation and SPICE models compatibility.

tomhajjar, 2022-11-23:
I  used to get the latest Qucs Windows release but they stopped posting them 3 years ago. I have a very old release of Qucs 0.0.20.

I have asked many times for access to the latest Qucs 0.0.20 release but no one will provide it. They seem to be off on a new tangent based on Gnucsator. I then asked for it and was ignored.....

ra3xdh, 2023-03-03:
I have just added INDQ and CAPQ devices. The attached screenshot illustrates its operation under Qucs-S+Ngspice. It should operate for AC, SP and TRAN analysis. These devices will be available since the next release v1.0.2.
[image]

ra3xdh, 2023-03-03:
And here is INDQ device: 
[image]

ra3xdh, 2023-03-04:
The attached screenshot shows a more complex example involving inductive coupling of INDQ devices.
[image]

ra3xdh, 2023-03-04:
Closing this as resolved. New devices will be available since v1.0.2 release.

closed 2023-03-04 by ra3xdh