ra3xdh/qucs_s #144: syntax error when creating component from spice library

tlatorre-uchicago, 2022-08-25:
I'm trying to create a component from the spice library for the AD8099 op amp here: https://www.analog.com/en/products/ad8099.html. I was able to successfully simulate the component in a circuit when I directly use the SpiceLibComp component, but when I try to create a subcircuit with ports for it and run the simulation I get the error:

```
Ngspice started...
Netlist line no. 0:
 Mismatch: 1 formal but 0 actual params.
gmin=$;
Netlist line no. 0:
Syntax error: letter [$]
Netlist line no. 0:
 Expression err: $
Netlist line no. 0:
 Formula() error.
      gmin=$;

ERROR: fatal error in ngspice, exit(1)
```

I did find a discussion about the problem here: https://www-linux-org-ru.translate.goog/news/opensource/16795684/page2?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp, but I'm relatively inexperienced with the software and don't fully understand all the solutions.

I tried downgrading ngspice to version 27 as suggested, but then got other errors when trying to simulate it.

Did I understand the conversation correctly that this is a bug introduced in newer versions of ngspice and not related to a problem in qucs-s?

tlatorre-uchicago, 2022-08-25:
Ah, ok realized this is because I named the schematic ad8099.sch which is also the name of the model. Renaming the schematic fixed the problem.

ra3xdh, 2022-08-26:
This may be related to #82 I suspect duplicate `.SUBCKT` entry somewhere in the auto-generated netlist. Since the recent versions the Ngspice doesn't allow duplicate subcircuit names. Unfortunately the error message is very cryptic and doesn't help to debug this issue. You may use `SpiceFile` device instead of `SpiceLibComp` it may be free from this issue but requires to copy every subcircuit to an individual netlist file.

holvo, 2022-08-26:
You may have a look here: http://s-audio.systems/blog/spicemodels/

ra3xdh, 2022-09-25:
No further complain since 1 month. Assume the issue has been resolved. Closing ticket.

closed 2022-09-25 by ra3xdh