ra3xdh/qucs_s #145: Use xdg base directory specification

moshiur-raj, 2022-08-26:
Currently qucs uses ~/.qucs . This contributes to littering the home directory. According to the xdg specification configs should be located at ~/.config , data should be located at ~/.local/share and cache should be located at ~/.cache .
Simply moving ~/.qucs to ~/.local/share/qucs would probably be the easier choice.

ra3xdh, 2022-08-27:
Yes, I am planning to move `.qucs` elsewhere but with the different purpose. The `$HOME/.qucs` directory doesn't hold config. The temporary files (netlist and log ) are located at `$HOME/.qucs/spice4qucs` (probably could be moved to .cache). The config file location is `$HOME/.config/qucs/qucs_s.conf` The `$HOME/.qucs` is used as the default location for projects and user library file. This is not good, because the hidden direcotory is not visible in file manger for default that may make difficult finding files. 

I propose the following layout for these directories:

*  `$HOME/.config/qucs/qucs_s.conf` config file. Default Qt applications config location.
*  `$HOME/cache/qucs` for holding temporary files like netlist and simulator data
* `$HOME/QucsWorkspace` default location for projects and user libraries. Not a hidden directory because need to visible from file manager by default.

moshiur-raj, 2022-08-27:
Qucs automatically creates .qucs in my home directory, even when I change the location in settings. Can you fix this too?

ra3xdh, 2022-08-27:
Yes, Qucs-S now creates default directory layout on the first start now. It is related to fix of #86. The usage of `QStandardPaths` instead of hardcoded paths may help to resolve this issue https://doc.qt.io/qt-5/qstandardpaths.html

ra3xdh, 2024-04-07:
The implementation of this provided by #658

closed 2024-04-08 by ra3xdh

moshiur-raj, 2024-05-04:
qucs still creates ~/.qucs at startup @ra3xdh .