ra3xdh/qucs_s #146: Passing params to subcircuit

tomhajjar, 2022-08-26:
What is the procedure to pass params to a sub circuit? Not having any luck. 

genericopa1.lib, "Ghost" parameters Pin generated and simulation fails

.subckt genopa1 in+ in- vcc vee out params: POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg
[image]

Edited genericopa1.lib -> genericopa1_mod.lib. Simulation works but no ability to alter parameters at "sch" level. 
.subckt genopa1 in+ in- vcc vee out 
+ params: POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg

[image]

[Passing params to subcircuit.zip](https://github.com/ra3xdh/qucs_s/files/9436248/Passing.params.to.subcircuit.zip)

tvrusso, 2022-08-26:
The parameters are passed from the top level circuit to the subcircuit by attaching additional properties to the element in the top level circuit, and by specifying which parameters may be passed in the subcircuit *symbol*.  Your subcircuit symbol doesn't define its input parameters, and where you instantiate that symbol in the main circuit doesn't have them added to the properties dialog.

I have an example of doing that in the Xyce_qucs-s_Examples precisely because it took me a while to figure out how to get it done.  It's example "05-NonlinearResistor".

[image]
[image]
[image]

tvrusso, 2022-08-26:
Note in this example I'm defining *default* values for the three parameters in the symbol.  These may be overridden by the main circuit when the properties are attached to the instance of the symbol when it is placed.  If they're left out, the defaults are used.  In this particular example, the top-level circuit specifies the three parameters even though it's just using their defaults.

tvrusso, 2022-08-26:
And just refreshing my memory here I played with the example.  You have to add the parameters to the subcircuit symbol *first*. 

They then just show up as properties on the symbol when it appears in schematics, and you can set them there.

tomhajjar, 2022-08-27:
This "old" method for passing parameters is no different than Qucs. It also works for Qucs-S however Qucs-S also allows one to enter params data directly as shown below. What's the point of doing this if this data isn't used? 

Still have an issue for libs that place params data on same line as .subckt

[image]

Params data is blank.... even though the data was entered.... 

[image]

tomhajjar, 2022-08-27:
The param data for the Qucs example quarz.sch is embedded in the ".ID" line. Many Qucs examples "ported" over to Qucs-S were never updated to reflect all of Qucs-S differences.
 
<.ID -40 34 ZQ "1=f=8863k=Crystal Frequency=" "1=Lq=0.01406=Crystal Inductance=" "1=Cs=6.5p=Crystal Capacitance=">

 The param data for the Qucs-S model I made is embedded in the "SpLib" line. The question is why is this data ignored. 

<SpLib X1 1 110 160 -26 81 0 0 "genericopa1_mod.lib" 1 "genopa1" 1 "auto" 0 "POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg" 1>

ra3xdh, 2022-08-27:
> enericopa1.lib, "Ghost" parameters Pin generated and simulation fails

Yes, it's a bug in `SpiceLibComp` device. The workaround is to manually edit the SPICE netlist and remove the `params:` keyword after the pin list. 

> What is the procedure to pass params to a sub circuit? Not having any luck.

The old `SpiceFile` device doesn't support passing parameters. I cannot fix this, because all existing circuits using this device will be broken. But passing parameters to SPICE subcircuits works as expected with new `SpiceLibComp`. The procedure is the same as described in documentation. Use the `Params` property for standalone SpiceLibComp. If you wrap `SpiceLibComp` into the subcircuit, then add parameters in the symbol editing mode (activated after pressing F9). Then pass parameters to SpiceLibComp. Attached screenshots illustrate this.

[image]
[image]
[image]
[image]

ra3xdh, 2022-08-27:
Also look at the attached schematics. Don't forget to correct the file paths before the simulation. 
[genopa.zip](https://github.com/ra3xdh/qucs_s/files/9437379/genopa.zip)

ra3xdh, 2022-08-27:
> The param data for the Qucs example quarz.sch is embedded in the ".ID" line. Many Qucs examples "ported" over to Qucs-S were never updated to reflect all of Qucs-S differences.

The Quartz example is made in a different way. It is combined from lumped devices and doesn't use external spice netlists. It works as expected with the actual Qucs-S+Ngspice version and doesn't need correction. But you may pass parameters to the subcircuit that use spice netlist as mentioned in my post above. You need to edit component ID in symbol editing mode (F9).

tomhajjar, 2022-08-27:
I'm still confused. The modified lib file removes "params:" from the .subckt line. No more ghost params pin.
.subckt genopa1 in+ in- vcc vee out params: POLE=20 GAIN=20k VOFFSET=5m ROUT=10 RIN=10Meg

Is the params info supposed to get automatically filled or do I have to go to the symbol edit mode and manually enter them?  

[image]

ra3xdh, 2022-08-27:
> In a future release, will the params info in the .subckt line get automatically filled into the SpiceLibComp Params window?

No, I am not inclined to fill this field automatically. The user can rely on default params or skip some parameters. There is no unified solution here.

> You manually added params into the SpiceLibComp Params window but used variables for GAIN=gain and VOFFSET=vofs. Is this a requirement to get parameters passed to the passed to the symbol editing mode? What about default values?

This procedure is required when you are wrapping `SpiceLibComp` in subcircuit and want to forward parameter to the nested subcircuit. You need to define the default values in the ID properties dialog (field "Default Value") in symbol editing mode. 
[image]

tomhajjar, 2022-08-27:
Sorry to keep asking questions, but isn't it confusing to have the ".ID" parameters used if you place a subcircuit.sch onto your main.sch but the "SpLib" parameters are used if you place a SpLibcomp onto main.sch? 

Wouldn't passing the "SpLib" parameters to the "end" schematic avoid confusion? For Spice, what is the purpose of using ".ID" to pass the parameters in stead of the SpLib" parameters? 

I have never placed a SpLibcomp onto my main.sch. I have only used them in my subcircuit.sch. Is this commonly done?

[image]
 
[image]

ra3xdh, 2022-08-27:
>  but isn't it confusing to have the ".ID" parameters used if you place a subcircuit.sch onto your main.sch but the "SpLib" parameters are used if you place a SpLibcomp onto main.sch?

The .ID approach is inherited from Qucs. If you have SpiceLibComp inside subcircuit you need to forward parameters to inner subcircuit. The only way is using the .ID The .ID has many advantages like parameter description that cannot be implemented using the SpiceLibComp. I cannot modify subcircuits to improve its compatibility with SPICE, because otherwise ALL schematics using subcircuits will be broken. The solution is to wrap SpiceLibComp in subcircuit. 

> I have never placed a SpLibcomp onto my main.sch. I have only used them in my subcircuit.sch. Is this commonly done?

Yes SpiceLibComp is designed for directly placing on schematic. You may also copy the SPICE `*.lib` file to the `$HOME/.qucs/user_lib` and drag and drop subcircuits and models from it. The subcircuit will be created with default rectangle symbol.

ra3xdh, 2022-09-22:
Assume the issue has been resolved. The found bug with `params:` keyword is fixed. The fix will be available in the next release. Closing this.

closed 2022-09-22 by ra3xdh

tomhajjar, 2022-10-30:
As a follow up, for passing parameters to work for a subcircuit, it seems you must put entries in two places as shown below.

Also the component lib file may need to be edited to remove the params: statement. 

[image]
[image]

[Op Amp Sub_Param_Pass_prj.zip](https://github.com/ra3xdh/qucs_s/files/9894840/Op.Amp.Sub_Param_Pass_prj.zip)