ra3xdh/qucs_s #147: Symbol generation and editing and available symbols

tomhajjar, 2022-08-30:
Any reason "Qucs" never had a Symbol editor? I have been "stealing" symbol sections out of other peoples schematics using a text editor or use symbols.sch from QucsStudio. Editing Qucs-S schematics using a text editor is not user friendly. 

On a side note, I have complained about the Elliptic Arc painting function over the years on the Qucs SourceForge forum to no avail. It is extremely frustrating to make good looking transformers and coils using Elliptic Arc. I always give up and steal an inductor from symbols.sch. I find it funny the transformers in Qucs-S only use straight lines.   

[symbols.zip](https://github.com/ra3xdh/qucs_s/files/9456341/symbols.zip)

tomhajjar, 2022-08-30:
Hmmmmm

https://sourceforge.net/p/qucs/mailman/message/35830246/

ra3xdh, 2022-08-31:
Currently the symbol may be edited only for schematic (entered by F9). But there is no editor for `*.sym` files. The `*.sym` files are relatively new features and were added in Qucs-S. Basic Qt4 Qucs doesn't support symbol files and use only subcircuits. The editing of `*.sym` files is technically possible, but was not implemented due to lack of developers and time resources. Probably symbol editor may be added in the future.

tomhajjar, 2022-08-31:
Maybe as a stop-gap provide groups of generic symbols in schematic form like I attached earlier. It sure speeds up subcircuit creation. I've been using this file for a long time. I plan on adding more symbols. Its time consuming to cut and paste symbols using a text editor. 

[image]

tomhajjar, 2023-09-28:
These schematics with various symbols speeds up creating devices and subcircuits. Just copy and paste. 

[Symbol_creation_prj.zip](https://github.com/ra3xdh/qucs_s/files/12754554/Symbol_creation_prj.zip)

ra3xdh, 2023-10-04:
@tomhajjar  I have uploaded your symbols into the `examples/symbols` subdirectory. I hope this will help other users to create their own subcircuits. Thanks for contribution!

tomhajjar, 2024-01-29:
Updated the symbol files.

[image]

[Symbol_creation_prj.zip](https://github.com/ra3xdh/qucs_s/files/14079253/Symbol_creation_prj.zip)

ra3xdh, 2024-04-23:
The symbol file editing feature implemented by #700

ra3xdh, 2024-05-03:
The solution provided by #700. Closing.

PR #700 (2024-05-03): Add symbol editor
Opening a PR following the discussion in #679

Recap:
- Symbol editor based on usual schematic editor
- Saving and opening detached symbols to and from *.sym files
- To save and open detached symbols, "Symbol only" filter should be selected in "Open file" dialog
- To start creating a symbol, switch to symbol editing mode
- Ports can be added and removed in symbol editing mode. If the symbol is then saved as *.sym file then all manually inserted ports remain. If it is saved as usual *.sch file, then all ports without a corresponding port in schematic will be removed.

May clos [... cut]
closed 2024-05-03 by ra3xdh