ra3xdh/qucs_s #148: Filter Synthesis tool s-parameter equation does not work

drewlio, 2022-09-08:
Qucs-s version 0.0.24
Ngspice 37, compiled with `set ngbehavior=ltpsa`

Filter synthesis tool produces a schematic with s-parameter simulation block and equation block, as below.
[image]

The equations are not correctly parsed by Ngspice, see ngspice output below.
```
Ngspice started...
Netlist line no. 4:
Undefined parameter [db]
Netlist line no. 4:
 Expression err: db(s[2,1])
Netlist line no. 4:
 Formula() error.
      .param dbs21={db(s[2,1])}
Netlist line no. 5:
Undefined parameter [db]
Netlist line no. 5:
 Expression err: db(s[1,1])
Netlist line no. 5:
 Formula() error.
      .param dbs11={db(s[1,1])}

ERROR: fatal error in ngspice, exit(1)

Note: Compatibility modes selected: ps lt a

Circuit: * qucs 0.0.24 /home/drew/.qucs/test2_prj/test2.sch

Simulation finished
```
A workaround is to delete the the auto-generated equation block and replace it with a Nutmeg equation where the simulation is set to `sp`
[image]
Then the simulation runs, and you can plot the variables defined in the nutmeg equation.
[image]
[image]

Additionally, it is puzzling that the dataset names are prefixed by `ac` and not `sp`. 
[image]

tomhajjar, 2022-09-08:
It works if you use the default Qucsator engine. 

For ngspice you use Nutmeg equations to get dB. 

For xyce Spice you don't need any equations for dB.

ra3xdh, 2022-09-09:
Yes, I am aware about this problem. I have added S paameter simulation with Ngspice/Xyce, but didn't update the filter synthesis utility. The solution may be the following:
* When Qucsator is selected generate legacy equations;
* For Ngspice add Nutmeg equations;
* For Xyce no equations, because Xyce doesn' support postprocessing;

Reagarding the `ac.` prefix for S-parameter data I cannot fix this. It comes from Ngspice itselt. The `ac.` means all frequency domain data including S-parameter simulation. The FFT simulation will also generate data with `ac.` prefix.

drewlio, 2022-09-09:
Another option for you to consider is a radio button in the Filter Synthesis tool where the user can select the equation type before copying the schematic. (but it's not a very elegant solution)

Thanks for the reply, and for all you do on this great software!

ra3xdh, 2022-09-09:
The substitution of the appropriate equation could be done automatically. This is similar to disabling of unsupported filter types #96

ra3xdh, 2022-09-10:
I have fixed the equations for auto-generated circuit by https://github.com/ra3xdh/qucs_s/commit/2111e526067d5c94d7592cdfa8c3245af904f079 . Nutmeg equations are inserted if Ngspice is selected as default simulator in the application settings. This fix will be available in the next release.

closed 2022-09-10 by ra3xdh