ra3xdh/qucs_s #152: AnalogueCM.lib issues

tomhajjar, 2022-09-11:
The AnalogueCM.lib library doesn't work anymore. Some type of path issue

ra3xdh, 2022-09-12:
This library uses XSPICE code models and requires installed GCC compiler and make utility for operation. Otherwise it will not work. The `AnanlogueCM.lib` is only library using Code models. It is not a problem for Linux platform but may have some difficulties for Windows. Do you have MINGW and Make installed and available in `$PATH`?

tomhajjar, 2022-09-12:
I believe I did a very long time ago on another PC. Not sure this library ever worked however. The library is not useful for me so it's not worthwhile trying make it work.

tomhajjar, 2022-09-17:
I guess you can add the MESFETCL1 component in the MESFET library to the list

tomhajjar, 2022-09-17:
[image]
[image]
[image]

[MESFET_Curtice_prj.zip](https://github.com/ra3xdh/qucs_s/files/9592686/MESFET_Curtice_prj.zip)

ra3xdh, 2022-09-18:
Yes, there is a bug with Qucs equations support. The parser doesn't recognize the `V1` voltage near the `&&` logical operator and doesn't convert it to the SPICE syntax. As a workaround you may to wrap the condition in parentheses. 
[image]
After this correction is applied, the circuit could be simulated without error. Also the equation `Eqn1` should be deleted because it duplicates the current probe.

ra3xdh, 2022-09-24:
I have added a fix for the EDD equations parser. Closing this.

closed 2022-09-24 by ra3xdh