ra3xdh/qucs_s #154: Deactivate doesn't work for Equations

tomhajjar, 2022-09-12:
In "porting" Qucs projects over to Qucs-S, it's helpful if Equations and components can be "Deactivated" instead of deleted. 

Qucs-S doesn't deactivate Equations and ngspice either crashes or flags errors. 

[image]

tomhajjar, 2022-09-12:
This project hasn't been ported to Qucs-S yet. I only replaced the Bias T, DC block and DC feed 

[Varactor Model_prj.zip](https://github.com/ra3xdh/qucs_s/files/9551397/Varactor.Model_prj.zip)

ra3xdh, 2022-09-13:
Yes, I confirm the bug with equations. I have also noticed another problem with SP simulation. It cannot process single point sweep properly.

ra3xdh, 2022-09-14:
I have fixed the equations problem, but the problem with single point simulation still remain.

closed 2022-09-21 by ra3xdh