ra3xdh/qucs_s #156: Strange behavior trying to simulate Varactor

tomhajjar, 2022-09-15:
I'm getting weird results trying to "port" a Qucs Varactor simulation over to Qucs-S. I want to use "Varactor_Model.sch" in circuits. I used the built in varactor model for debugging.

Like for S-parameters, AC simulation for a "constant" fixed frequency doesn't work. Forced to have 2 frequencies. 

As shown below, the data for "Component Impedance_I.sch" is as expected. Simple LCR meter. 

The data for "Varactor_Impedance_I.sch" is garbage. I tried disabling various componenst 

[image]

[image]

[Varactor Model_prj.zip](https://github.com/ra3xdh/qucs_s/files/9576510/Varactor.Model_prj.zip)

ra3xdh, 2022-09-15:
It seems this example cannot be ported directly to Ngspice. The SPICE has no single point simulation type. It is exclusively Qucsator invention. The SPICE standard supports only sweep simulation. As the simulation result Ngspice always produces not a single point but a vector. See chapter 17 of the Ngspice manual https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-23509 . The `frequency` variable is also not valid for Ngspice. You should look at the chapter 17.2 of the Ngspice manual and rewrite equations using Ngspice notation. Then it will probably work.

tomhajjar, 2022-09-15:
If you compare the two schematics, after disabling some components, they are equivalent. Why does the first work and not the second?

ra3xdh, 2022-09-15:
Disabling of parameter sweep doesn't work properly. If I delete it manually the capacitance is calculated fine.

tomhajjar, 2022-09-15:
If start/stop are the same frequency or you sweep only 2 points, you only get data for the first point.

You get Tabular data for 1 or more data points. You get a Cartesian plot for 3 or more data points. 

Could you not send ngspice the same freq twice (Start=Stop) when in Constant mode? 

Qucs puts "0 nan" if Start=Stop

[image]

[image]

tomhajjar, 2022-09-16:
Vadim

My goal was to make a generic varactor model with parasitics like done in Qucs/QucStudio. I was able to get ngspice to generate CV data in Tabular form but not as a plot of C versus V. The caveat was I was forced to put the varactor parameters (Cj0, M, Vj, Bv) directly into the diode model used in the subcircuit, Varactor_Model_mod.sch. I could not figure out a way to get Qucs-S to pass the parameters to the diode in the subcircuit, Varactor_Model.sch. 

Is there a reason Qucs/QucStudio will pass parameters to any device like semiconductors in a subcircuit but but Qucs-S won't? I realize this isn't something many would need.

[image]

[Varactor Model_prj.zip](https://github.com/ra3xdh/qucs_s/files/9586827/Varactor.Model_prj.zip)

tomhajjar, 2022-09-17:
I found an online ngspice example for plotting varactor C versus V for a fixed AC frequency. 

https://sourceforge.net/p/ngspice/discussion/120972/thread/cd75976f/
[Varactor CV ngspice.zip](https://github.com/ra3xdh/qucs_s/files/9590774/Varactor.CV.ngspice.zip)

ra3xdh, 2022-09-17:
> Is there a reason Qucs/QucStudio will pass parameters to any device like semiconductors in a subcircuit but but Qucs-S won't?

It's a bug. Qucs-S cannot recognize variable in model parameter and substitutes zero. I will provide a fix soon.

tomhajjar, 2022-09-17:
Will Qucs-S implement PlotVs?

ra3xdh, 2022-09-17:
The `PlotVs` was implemented long time ago. It is available from diagram dialog, but not as function. 
[image]

tomhajjar, 2022-09-17:
Is the CV plot blank due to the Auto ranging issue?

Why is the DC sweep not 0, 1, 2...10? It's 0, .90909, 1.81818...10?

[Varactor Model_prj.zip](https://github.com/ra3xdh/qucs_s/files/9591987/Varactor.Model_prj.zip)

ra3xdh, 2022-09-17:
No, is not related to auto-ranging issues of #133 Linux version is affected too. I need to make more investigations to determine the reason of blank plots. Look like inconsistent vector dimension of X/Y axis somewhere.

ra3xdh, 2022-09-17:
This gives a bit more light on this. The root of this is the fact that single point simulation in SPICE is a hack. The every single point simulation produce a set of 1x1 point plot.  If you set curve type `stars` instead of `solid line` you will see a plot instead of blank diagram.  But the `PlotVs` still fail.  
[image]

ra3xdh, 2022-09-17:
You will be able to see the C-V plot using the 3-D diagram.
[image]

ra3xdh, 2022-09-20:
See also my comments here: https://github.com/ra3xdh/qucs_s/issues/161#issuecomment-1252103707 The #161 is related to this.

tomhajjar, 2022-09-20:
I updated my files to get around the known issues. I will update them as fixes are implemented.
I added measurement of CV using S-parameters. P2 must be terminated for ngspice. The C calculations ignore model parasitics.

Did you discover why "V1" is 0, .909, 1.82.. instead of 0, 1, 2...?

[image]

[Varactor Model_prj.zip](https://github.com/ra3xdh/qucs_s/files/9607551/Varactor.Model_prj.zip)

ra3xdh, 2022-09-20:
> Did you discover why "V1" is 0, .909, 1.82.. instead of 0, 1, 2...?

The step for netlist is computed as `(stop-start)/points=(10-0)/11` The step that shown in the properties dialog is different.

tomhajjar, 2022-09-28:
I tried every combination of Star, Stop, Points, yet I can't get a plot with a starting Vgrid=0. 

If I try to Start=-5, Stop=1, Points=7, Vgrid=1, 0.143. -0.714...-5

I tried making Start=0, Stop=-5, Points=6 but ngspice does nothing. 

Is this a bug?

[image]
[image]

tomhajjar, 2022-09-28:
I simulated some BSIM3v3 files in ngspice and there was no problem plotting Ids vs Vgs when Vgs =0, 1....

[image]

[BSIM3v3.zip](https://github.com/ra3xdh/qucs_s/files/9666270/BSIM3v3.zip)

ra3xdh, 2023-06-04:
Look for the solution of the CV-curve plotting problem in #277. This currently may be resolved using some tricks with Nutmeg scripting. No application fix is required. I am closing this ticket.

closed 2023-06-04 by ra3xdh