ra3xdh/qucs_s #158: New 3 terminal MOSFETs

tomhajjar, 2022-09-17:
Shouldn't two of the new "3 pin" MOSFETs use the enhancement mode symbol? Is it redundant to offer 3/4 pin selections for these depletion mode devices when four, 4 pin depletion mode devices exist as well? Am I missing something?

What is your definition of a Spice "X" device versus an "M" device? 

[image]

ra3xdh, 2023-02-03:
> Shouldn't two of the new "3 pin" MOSFETs use the enhancement mode symbol?

I don't see a big issue here. Some other CADs are also using common symbol for both enhancement and depletion mode MOSFET devices. 

> What is the difference between the Spice "X" device versus an "M" device when both offer the same Properties?

The "X" device is mapped to SPICE entry `X` letter representing subcircuits. Some of PDKs for microelectronics design provide MOSFET models wrapped in subcircuit. The X MOSFET device is needed to support such libraries from PDKs.

ra3xdh, 2023-02-04:
Assume this is resolved. Closing ticket.

closed 2023-02-04 by ra3xdh