ra3xdh/qucs_s #160: BSIM example projects

tomhajjar, 2022-09-19:
Gmail and Yahoo blocked me from sending you this attachment for the email I sent earlier.

Email text:
I did more work on the BSIM example projects. Strangely I cannot get 2 of the pMOS 3/4 devices to generate curves using ngspice but the pMOS 3/4 devices do work in the other test circuits. The nMOS 3/4 devices work in all cases.

The following 2 won't work

BSIM3v3_pMOS_Ids_Vgs.sch
BSIM4pMOS_Ids_Vgs.sch

The BSIM6 simulations using ngspice are very questionable.I suspect the bsim6v1mos.lib I used is garbage. This is a library was found in an old Qucs example. 

The BSIM6 data using xyce looks correct. The library used was BSIMBULK.lib. 

[BSIM_projects.zip](https://github.com/ra3xdh/qucs_s/files/9600441/BSIM_projects.zip)

ra3xdh, 2022-09-19:
> The BSIM6 simulations using ngspice are very questionable. 

Ngspice doesn't include BSIM6 models, but contain examples for BSIM6. It may be misleading. The simulation result will be incorrect if one attempts to use BSIM6 device. But XYCE has a full support of the BSIM6 device. 

> Strangely I cannot get 2 of the pMOS 3/4 devices to generate curves using ngspice but the pMOS 3/4 devices do work in the other test circuits

Yes, I can reproduce this issue. It is not related to PMOS device itself. The auto-generated  Nutmeg script is incorrect for negative voltage values. You may flip the sources as a workaround. After that you will obtain correct IV-curves as shown. 
[image]

tomhajjar, 2022-09-19:
ngspice gets "stuck" trying to simulate BSIM4pMOS_Ids_Vgs.sch

Used Qucs-S 1.0.0 and 0.0.24

tomhajjar, 2022-09-19:
BSIM6v1_idvd_pmos.sch works with the same source polarities of the failed
BSIM3/4 analysis. It's got to be something in addition to the Nutmeg script

ra3xdh, 2022-09-19:
Yes, I can reproduce the problem with BSIM4. But it is not related to Qucs-S. The Ngspice reports about convergence error and hangs. You can see the log after pressing the Stop button in the dialog window. You are trying to simulate the device outside its operation limits. Breakdown point has achieved during DC sweep that causes Ngspice convergence failure. After one corrects voltage everything works as expected. 
[image]

ra3xdh, 2022-09-19:
> BSIM6v1_idvd_pmos.sch works with the same source polarities of the failed

BSIM6 simulations with Ngspice are all incorrect. Ngspice doesn't contain the BSIM6 device and simulate this circuit with the default model. It may converge, but the result is totally wrong.

ra3xdh, 2022-10-04:
The reported issue with negative numbers has been fixed on `current` branch. Closing this.

tomhajjar, 2022-10-11:
Holger posted BSIM info that I used. I guess the version of ngspice I am
using is not supported.

https://sourceforge.net/p/ngspice/ngspice/ci/adms-experiment-3/tree/tests/