ra3xdh/qucs_s #161: Problem with CV curve displaying using Xyce simulator (one-point AC analysis)

georgtree, 2022-09-20:
Hello, the issue is contained in attaching schematic file (I had to change .sch to .txt for being able to upload file).
Step to reproduce:
1) Run simulation with Xyce simulator
2) Try to display output data {IM(v2)} on 2D cartesian graph using function PlotVs. versus {V1} Then you will get error, but data could be displayed using Tabular Data without PlotVs.

I assume that problem is due to one-point AC simulation, but how I could plot CV graph?
[ac_diode.txt](https://github.com/ra3xdh/qucs_s/files/9605579/ac_diode.txt)

georgtree, 2022-09-20:
I assume that it's similar issue to this https://github.com/ra3xdh/qucs_s/issues/156

ra3xdh, 2022-09-20:
Yes, the issue is similar to #156 The XYCE produces a set of 1x1 point plots after the simulation. You can see them if set line type as `stars` instead of `solid line` in diagram properties. The plotVs emulation has limitation that prevents from getting a CV-plot. The plot vs accepts only two vectors of the same length like X-Y oscilloscope mode. For C-V it will not work. Because C is not a vector but a set of 1x1 vectors. Only workaround is to use 3-D plot. It will allow you to see the C-V curve. 
[image]

ra3xdh, 2022-09-20:
I will look how can I fix the PlotVs emulation to make it accept combined vectors. I have added todo list to this issue.

ra3xdh, 2023-06-04:
The problem with CV-curve plotting has been resolved for Ngspice. Look at #277. It's unclear how to use the similar approach for XYCE.

ra3xdh, 2025-01-03:
Still be unclear how to fix CV-curve simulation with XYCE. Closing, because no solution provided by community after two years.

closed 2025-01-03 by ra3xdh