ra3xdh/qucs_s #162: Make revision of the examples

ra3xdh, 2022-09-20:
The examples from `examples` directory were made for Qucsator simulation kernel and may be incompatible with Qucs-S+Ngspice. Like `examples/resonance.sch` will not work work with Ngspice and produce cryptic error message. This may be misleading for new users. It's need to review all examples and move inconpatible with Ngspice to special `qucsator` subdirectory.   Another task is to add more examples.

- [ ] Move Ngspice-incompatible examples elsewhere
- [ ] Add new examples

tomhajjar, 2022-09-21:
I've segregated the examples into separate "schematic" folders under separate start folders for Qucs, Qucs-s and QucsStudio. "prj" folders are also segregated by qucs version. I have tested almost every schematic and project.  

[image]

tomhajjar, 2022-09-21:
One of the issues I see with the examples is the non-descriptive names used for schematics. Unless there is a directory, no user is going to know what many of the circuits are. 

When I look for help trying to do something, I often use Grep looking for keywords in the examples folder. I then look at the example schematic hoping it will help. Since most of the examples are for "Qucs", it isn't as helpful as it could be.

tomhajjar, 2022-09-21:
When "porting" Qucs schematics to Qucs-S, I'm slowed down by Qucs-S not generating any error messages. I don't know ngspice well enough to convert the two Parameter Sweeps used in Qucs to work for Qucs-S. Getting no error messages doesn't help.

[image]
[image]

[schematics-S.zip](https://github.com/ra3xdh/qucs_s/files/9618872/schematics-S.zip)

tomhajjar, 2022-09-21:
There are a number of examples that use BJT with Substrate. Qucs-S doesn't support this device but I believe ngspice and xyce does.

tomhajjar, 2022-09-22:
First cut at the revised examples. ngspice and xyce folders to follow

[schematics-S.zip](https://github.com/ra3xdh/qucs_s/files/9621906/schematics-S.zip)

tomhajjar, 2022-09-22:
ngspice and xyce folders
[ngspice_xyce_schematics.zip](https://github.com/ra3xdh/qucs_s/files/9621907/ngspice_xyce_schematics.zip)

ra3xdh, 2022-09-22:
Thank you @tomhajjar  I will sort the examples in the next few days and leave feedback. One remark: the `*.dat.*` files (datasets) must not be included in the Qucs-S distribution to avoid the bloating of the package. Only `*.sch` and `*.dpl` files should be indexed by git.

tomhajjar, 2022-09-22:
I redid the xyce folder and added some files.

Qucs-S 1.0.0 still has issues with autoranging for xyce plots.

[xyce.zip](https://github.com/ra3xdh/qucs_s/files/9627723/xyce.zip)

ra3xdh, 2022-09-22:
> When "porting" Qucs schematics to Qucs-S, I'm slowed down by Qucs-S not generating any error messag

Qucs-S doesn't generate error messages. It only shows the error messages from simulation kernel. Unfortunately Ngspice doesn't generate very verbose error messages and often it show only `no .plot no .print line` and nothing more. I cannot do anything with this. I can only prevent user from insertion of the incompatible device or equation. 

> Qucs can alter parameters of multiple components/sources with the same Parameter Sweep.

It is Ngspice generating no error messages but not Qucs-S. Ngspice doesn't support parameter sweep. And auto-generated Nutmeg script that emulates parameter sweep doesn't support sweeping two devices simultaneously. This schematic could be ported only using manual written Nutmeg scripting. More that two Parameter Sweep simulations referencing the same simulation is not allowed. I can perform the following actions to resolve this problem:

* Implement sweeping of more than one device if the device list is given like `R1;R2`
* Show warning if more that one Par.Sweep is referencing the same simulation

ra3xdh, 2022-09-22:
> Qucs-S 1.0.0 still has issues with autoranging for xyce plots.

I didn't fix the auto-ranging issue #133 yet. Only 64-bit Windows version is affected.

ra3xdh, 2022-09-22:
> There are a number of examples that use BJT with Substrate.

I will add these devices to Qucs-S soon. I have created a dedicated issue #164 .

tomhajjar, 2022-09-22:
Are you rethinking having so many depletion mode fet symbols and no enhancement mode ones?

ra3xdh, 2022-09-22:
I don't see any big issue with MOSFET look. The symbol view doesn't affect MOSFET operation, but only the component look. Many software use the common symbol for all MOSFETs. I can redraw the symbol, but it's a low priority task in my list.

tomhajjar, 2022-09-22:
What is your reasoning for having X and M devices in the Main dock when X and M can be changed and the parameters are the same?

[image]

ra3xdh, 2022-09-22:
These devices have different SPICE letter. The `X` device is represented as subcircuit and `M` device as represented as MOSFET device. PDK for microelectronics often provide libraries containing MOSFET models where MOSFET is represented as subcircuit. It make sense to have a special device for them.

tomhajjar, 2022-09-22:
FYI
single_phase_transformer.sch
Transformer_opposite_coupling.sch
Transformer_positive_coupling.sch

All had the same issue. Subcircuits had errors L=L1=L1 when they should be L=L1 ...etc

KAGVa, 2022-10-01:
> 

Hi, when you use the param_sweep.cpp and param_sweep.h from PR#169 you should now be able to sweep multiple parameters in a single sweep (only one type allowed), the syntax is
Param=R1;R2;R3
split with ";" - if you use different parameters like R1;C1 I don't know what happens (I haven't tried that)

ra3xdh, 2022-10-19:
I have just pushed the revised examples tree to the specific branch https://github.com/ra3xdh/qucs_s/tree/revised_examples I am planning to merge this into the `current` branch soon. The legacy examples are moved to the `examples/qucsator` subdirectory now.

ra3xdh, 2022-10-19:
> All had the same issue. Subcircuits had errors L=L1=L1 when they should be L=L1 ...etc

I have fixed this example by switching transformers to the library implementation from `Transformer.lib` These subcircuits were made when the library was not available, but the example was not updated later.

ra3xdh, 2022-11-25:
The Qucs v1.0.0 contains the revised examples. The schematics were copied in the new tree. The binary files has been deleted from Git. Thanks @tomhajjar  for help! It remains a question regarding the inclusion of the examples from here https://github.com/tvrusso/Xyce_qucs-s_Examples/ But this issue may be closed as resolved.

closed 2022-11-25 by ra3xdh