ra3xdh/qucs_s #163: Qucs-S closes abruptly

tomhajjar, 2022-09-21:
During some vacuum tube amplifier modeling, the tube model 6V6.sch based on 6V6.lib is causing Qucs-S 1.0.0 and 0.0.24 to abruptly close without generating any files when I try to analyze it with "6V6 DC Sweep.sch".   

[Tube_prj.zip](https://github.com/ra3xdh/qucs_s/files/9612059/Tube_prj.zip)

tomhajjar, 2022-09-21:
I was doing another comparison to LTSpice. That requires me to use the same Tube Spice models as the original analysis. People on diyAudio have their own preference for tube models. There has to be a dozen 12AX7 models alone....

[Tube_6CA7_Push_Pull_prj.zip](https://github.com/ra3xdh/qucs_s/files/9618712/Tube_6CA7_Push_Pull_prj.zip)

ra3xdh, 2022-09-25:
The crash has been fixed on `current` branch. The Qucs-S will substitute extension from the list only if no extension is given for the SPICE netlist. Always preserve original file extension.

closed 2022-09-25 by ra3xdh