ra3xdh/qucs_s #165: Failed to start simulator!

foobar1994, 2022-09-22:
I am using Gentoo GNU/Linux with ngspice-36:
```
/usr/bin/ngspice --version
******
** ngspice-36 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2020, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Thu Sep 22 18:28:56 UTC 2022
******
```
Ngspice executable location is set to "/usr/bin/ngspice" and default simulator is Ngspice. Every time I click "Simulate", I get a pop-up saying "Failed to start simulator!", with no further information. The log file does also not contain any information.

This proplem occurs with the latest AppImage as well as with a fresh build of qucs_s from source. I have also build and installed ngspice-37 manually, but this has also no effect.

This occurs with any project (example or new project).

ra3xdh, 2022-09-23:
I don't observe this issue on my Linux installation on Ubuntu. It look like the issue #86 Please check if the Ngspice working directory is writable in the application settings. The #86 has been fixed in the latest release. Old settings may remain if you tried older Qucs-S packages before.

ra3xdh, 2022-09-23:
Also try removing config file `$HOME/.config/qucs/qucs_s.conf` Make backup of this file before remove.

foobar1994, 2022-09-25:
Thank you for your reply, I followed your steps and it is now working as expected.

closed 2022-09-25 by foobar1994

ra3xdh, 2022-11-09:
Без схемы Qucs-S не запустит моделирование. Обязательно требуется валидная схема. Если ошибка остаётся, то следует проверить настройки путей в *Simulation->Select default simulator*

Neptun78rus, 2022-11-09:
cхема есть просто окошко закрыло
[image]
с путями тоже порядок
[image]
ngspice есть и отдельно запускается
[image]

ra3xdh, 2022-11-09:
У себя нигде не могу воспроизвести этот баг. В ближайшее время попробую установить Qucs-S на свежую виртуалку. Нет ли каких-то странных прав доступа на домашнем каталоге и на каталоге `$HOME/.qucs`?

ra3xdh, 2022-11-09:
И какой это дистрибутив?

Neptun78rus, 2022-11-09:
Это sparkylinux

ra3xdh, 2022-11-09:
Сейчас попробовал создать на виртуалке с Ubunutu создать новый профиль пользователя и запустить оттуда Qucs-S+Ngspice -- симуляция работает. Не могу воспроизвести баг. Попробую поставить в виртуалку SparkyLinux и посмотреть, что там происходит. Возможно там есть особенности, хотя по идее всё должно работать. Это же модифицированный Debian.

Neptun78rus, 2022-11-09:
drwxr-xr-x   3 root root       4096 окт 12 10:13 home
drwxr-xr-x 26 es7 es7 4096 ноя  9 20:07 es7
[image]

ra3xdh, 2022-11-09:
С правами вроде всё нормально. Нужно смотреть на виртуалке, что там происходит.

Neptun78rus, 2022-11-09:
https://sparkylinux.org/download/stable/
x64 lxqt
[image]

Neptun78rus, 2022-11-09:
я вначале в ручную поставить там cmake ругался вроде тоже на ngspice
а потом нашел под дебиан скрипт, и поставил, прога запустилась, а симулятор не запускает

ra3xdh, 2022-11-09:
> cхема есть просто окошко закрыло

Только эту схему пробовали? На этом скриншоте с двумя резисторами на схеме отсутсвует симуляция. Её нужно брать с панели в левой части. Следует добавить AC или TRAN analysis. Одиночная DC теперь не требуется и не имеет эффекта. Также см. туториал https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf Работают ли схемы из `/usr/share/qucs-s/examples`? Их нужно сначала копировать в место, где есть права на запись. 

> я вначале в ручную поставить там cmake ругался вроде тоже на ngspice

CMake не проверяет наличие ngspice, и он для сборки не требуется. Он ругался на что-то ещё.

Neptun78rus, 2022-11-09:
Мой косяк, я не разобрался  с ходу, оказалось  в examples все работает, просто меня сбило с толку сообщение -- No simulation found я подумал что симулятора не находит, сори, с англиским трабл..(

vitperov, 2024-02-17:
О! Это взрыв мозга!
Да, действительно с примерами всё работает! Но это так не очевидно! Когда вылезает ошибка ```
"No simulation found. Please add at least one simulation!
Ngspice error..." _ 
```