ra3xdh/qucs_s #166: Vacuum tube library and ngspice analysis issues

tomhajjar, 2022-09-22:
Triodes are named "VL" while Pentodes are named "Tube". Should they all be "VL" or something else?

Tube Models_prj contains sch and cir files I found online. They are the same as the built in library Tubes.lib. I have cleaned them up. They may be useful for making new tube subcircuits. The cir files and Tubes.lib don't state which pins are A, G or K. It slowed down making the models.

I have spent a while trying to get a tube amp that works in LTSpice to work with ngspice. So far no luck. ngspice generates no messages or data. 

6CA7_push_pull.sch uses the original models from diyAudio for the LTSpice analysis

6CA7_push_pull_lib.sch use the built in library Tubes.lib

[image]
[Tube modeling.zip](https://github.com/ra3xdh/qucs_s/files/9629516/Tube.modeling.zip)

ra3xdh, 2022-09-24:
> Should they all be "VL" or something else?

I will correct the library to provide an unified component name for all tubes (VL).

> I have spent a while trying to get a tube amp that works in LTSpice to work with ngspice

It is not a library error. You have a floating node outside the schematic field. You can make it visible if apply `View->View all` command. Also it is visible in the schematic XML file and located somewhere far away on the schematic sheet. After I copy schematic to the new sheet everything starts to operate. 
[image]

tomhajjar, 2022-09-24:
The "floating" node was caused by using Qucsconv from QucsStudio to convert the original LTSpice netlist to a Qucs-S schematic. Qucsconv doesn't exist in Qucs-S and the version in Qucs is horrible. I've converted many netlists to "Qucs" doing this in the past usually to make subcircuits. I found it useful to convert LTSpice files to Qucs. Unfortunately the file format of QucsStudio has multiple differences like sources rotated 90 degrees that cause floating nodes in Qucs. I usually find all of them but in this case it they ended up invisible far off sheet. Qucs-S renames nets on the fly and this seems to cause some nets to become invisible. Not getting any error messages for floating nodes is problamatic. 

Is there any consideration to reintroduce Qucsconv back into Qucs-S? Of course a version of Qucsconv that actually works. ;-)

ra3xdh, 2022-09-24:
> Is there any consideration of reintroducing Qucsconv back into Qucs-S?

It is possible only to backport qucsconv from the latest state of Qt4 Qucs git. The qucsconv utility from Qucs was not updated a long time ago and doesn't actually work. Qucsconv can only convert the `.MODEL` entries that is little help, because I implement a direct support of the SPICE models for Qucs-S. It cannot convert SPICE netlist to schematic and libraries and doesn't help user. Inclusion of this qucsconv version to Qucs-S is a waste of time. Also `qucsator` recently became a separate package. See https://github.com/qucs/qucsator . And qucsconv utliity is part of qucsator package. This may lead to the package conflict if someone will provide a Linux package for qucsator simulator. 

> Of course a version of Qucsconv that actually works.

The source code of the `qucsconv` utlity from QucsStudio is available. But I cannot fork these sources because the license is not specified by author. QucsStudio package implies some non-free license. And inclusion of this qucsconv version to another package may be considered as the copyright violation.

tomhajjar, 2022-09-24:
One way to increase the user base of Qucs-S is to make the transition from LTSpice to Qucs-S less painful. I'm converting select files to confirm accuracy of analysis. No one wants to hand enter a large LTSpice schematic into Qucs-S. Being able to import LTSpice schematics would help. I use a program that converts netlists to LTSpice schematics, SchBuilder.exe. I use Qucsconv to do similar for Qucs-S.

https://ltwiki.org/files/adventures_with_analog/SchBuilder/

ra3xdh, 2022-09-24:
> One way to increase the user base of Qucs-S is to make the transition from LTSpice to Qucs-S less painful

This question was asked before See #113 The LTspice has text format for the `*.asc` and `*.asy` files. So adding its support is technically possible.  But the import of the LTspice schematic may be a little help because of difference in system primitive devices library that should be taken into account. The adding support of the LTspice file may be time consuming task, so I don't consider its implementation in the near future. I have development tasks with higher priority like adding digital devices and implementing HF devices. Also the support of proprietary file formats without a permission from the copyright holder may infringe some legal issues. I may mistake in this subject.

tomhajjar, 2022-09-24:
Agree on all points. Adding more digital and RF is a priority. S-parameter file support should be a high priority.

tomhajjar, 2022-09-24:
I finally got the Qucs-S simulation to agree with LTSpice. 

6CA7_push_pull.sch uses the original models from diyAudio for the LTSpice analysis

6CA7_push_pull_lib.sch use the built in library Tubes.lib

[Tube_6CA7_Push_Pull_prj.zip](https://github.com/ra3xdh/qucs_s/files/9639124/Tube_6CA7_Push_Pull_prj.zip)

ra3xdh, 2022-10-19:
The reported issue with default component name for `Tubes.lib` has been fixed. Closing this ticked.

closed 2022-10-19 by ra3xdh