ra3xdh/qucs_s #171: "current" branch will not build, missing "qucslib_common.h"

tvrusso, 2022-10-07:
I just tried to build the code off of the current branch today and it failed.  qucs/qucs.cpp includes "qucslib_common.h" but that file no longer exists in the source tree.  I tried just removing the include, but there were undeclared identifiers and other issues, and other files also include qucslib_common.h.

ra3xdh, 2022-10-07:
Hello Tom, I forgot to commit `qucslib_common.h` after removing the `qucs-lib` tool #139. I have just restored the missing file. Please try to compile again. I have recently added CMake build test on the `master` branch to prevent such issues in the future https://github.com/ra3xdh/qucs_s/actions/workflows/cmake.yml This will be available for all branches after merging `master` and `current`

tvrusso, 2022-10-07:
Thanks, Vadim.  Yes, it is fine now.

I had just created a pull request while you were fixing it, but you beat me to the punch.

closed 2022-10-07 by tvrusso