ra3xdh/qucs_s #173: Multiple issues trying to port Qucs project to Qucs-S

tomhajjar, 2022-10-27:
I ran into many problems trying to port a Qucs current mirror project over to Qucs-S. My comments are for current_mirror_bjt.sch but the same problems exist for all the files. 

If a voltage/current probe is added to T2 collector, i(v1) and i(v2) currents and net voltages become unavailable for plots. 

If a net Vct2 is added to T2 collector, i(v1) and i(v2) currents and net voltages become unavailable for plots. 

Deactivating a voltage/current probe doesn't work. I have to delete them to get net voltages and currents to re-appear. 

I have tried every variation using the Nutmeg deriv function to replace the Qucs diff function to calculate Rout without success. I am missing something. The file mirrorbip_mod.sch shows my attempts. 

[image]
[image]
[image]

[Mirrors_prj.zip](https://github.com/ra3xdh/qucs_s/files/9880264/Mirrors_prj.zip)

tomhajjar, 2022-10-27:
Qucs current mirror variations files I recently edited.

[Mirrors_Qucs_prj.zip](https://github.com/ra3xdh/qucs_s/files/9880413/Mirrors_Qucs_prj.zip)

ra3xdh, 2022-10-27:
Hello Tom,

here are some explanations 
> If a voltage/current probe is added to T2 collector, i(v1) and i(v2) currents and net voltages become unavailable for plots.

You rely on undefined behavior in this simulation. The implicit names are used in the plot. If no named nodes specified in the schematic, the auto-generated `write` statement in the netlist will write **all** nodes to the dataset. If you add the voltage probe here, the dataset will contain only named nodes and probes. Probably disabling the probe doesn't work properly here. The solution is always name nodes explicitly with *Insert->Wire label* or add explicit probe and rely only on named nodes and probe variables in the equations to avoid such issues. 

> I have tried every variation using the Nutmeg deriv function to replace the Qucs diff function to calculate Rout without success.

There is no direct replacement for Qucs `diff()` function in Ngspice. You need to use a construction with `vecd()` function instead to get the `dV/dI` vector. See attached screenshot.

 
[image]

tomhajjar, 2022-10-27:
I added Ic1, Ic2 and flipped direction. What Ro calculation do you think is correct? They don't match.

[image]
[image]

[Mirrorbip.zip](https://github.com/ra3xdh/qucs_s/files/9881805/Mirrorbip.zip)

tomhajjar, 2022-10-27:
When I opened the identical sch and dat files generated by Qucs in Qucs-S, the plot of Rout in Qucs-S is scaled differently. I have not done any simulation in Qucs-S, I just opened the file. Notice the Equation files are the same. I have not edited this file to work with ngspice. Why would Rout be scaled differently when the two drain currents are scaled properly?

[image]
[image]

ra3xdh, 2022-10-28:
> When I opened the identical sch and dat files generated by Qucs in Qucs-S, the plot of Rout in Qucs-S is scaled differently

I cannot reproduce this bug on Linux platform. The values for Ron are identical for both Qucs and Qucs-S. Inspecting the `dat` file I can see the value for `Ron` approx. `1e14`. The Qucs-0.0.20 reads the dataset wrong.  So, this problem is not related to Qucs-S. Maybe this bug appears because the Qt4 library is deprecated and may operate wrong on modern platforms.   

[image]

ra3xdh, 2022-10-28:
> What Ro calculation do you think is correct?

I have tested the Ngspice approach with the diode series resistance. The equation `vecd(v)/vecd(i)` gives the correct result for forward resistance. 

[image]

ra3xdh, 2022-11-25:
Assume the questions raised during the porting Qucs schematics to Qucs-S has been resolved. Closing the ticket.

closed 2022-11-25 by ra3xdh