ra3xdh/qucs_s #176: How to disable Parameter Sweep; How to alter two parameters

tomhajjar, 2022-10-31:
Is there a reason a Parameter Sweep cannot be disabled? I have to delete the two sweeps to get ngspice to work.

A responder to one of the Issues mentioned that more than one parameter can be changed by specifying the parameters as R1;R2. 
I cannot find the comment to confirm. Is there a way to do it? ";" cannot be used in the parameter window

[image]
[diff1.zip](https://github.com/ra3xdh/qucs_s/files/9897627/diff1.zip)

ra3xdh, 2022-10-31:
> Is there a reason a Parameter Sweep cannot be disabled?

Yes, I confirm this bug. Disabling parameter sweep doesn't work properly. 

> I cannot find the comment to confirm. Is there a way to do it? ";" 

This feature was added by #169 and merged in `master`. Yes, I confirm the problem with semicolon symbol in the input field. Some validator is installed for this dialog that prevents from entering semicolon symbol. As the workaround you may enter the parameter name directly on schematic. Just click on the component property and it become editable.

ra3xdh, 2022-10-31:
After applying corrections the schematic could be simulated without error:
[image]

ra3xdh, 2022-10-31:
I have fixed this bug on the `current` branch. There was a validator for Par.Sweep property holding the component name. The disabling of the Par.Sweep also should work properly now.

tomhajjar, 2022-10-31:
I look forward to a newer Windows release so I can do more testing.

"As the workaround you may enter the parameter name directly on schematic. Just click on the component property and it become editable."

I don't understand this statement.

ra3xdh, 2022-10-31:
> I don't understand this statement.

Al component properties are clickable. After you click the component property, you may edit it. Then press Enter to apply changes. This may be done not only for simulations but for all components. 
[image]

tomhajjar, 2022-10-31:
A "feature" I never knew about. I only copied text from this field. 

Wouldn't this feature be more obvious if the editable field turned a different color like white?

ra3xdh, 2022-11-25:
Closing issue as resolved.

closed 2022-11-25 by ra3xdh