ra3xdh/qucs_s #177: Limitations changing the axis label name

BatchDrake, 2022-10-31:
Hi,

I've just tested Qucs-S 1.0.0 and it looks great, thanks for this release! I am using Debian GNU/Linux (Sid) on a x86-64 machine.

The first thing I tried was the S-parameter simulation and, while the results are correct, I find difficult to adjust the labels of the Cartesian plot. 

In particular, if I want to overlay the plots for S11 and S21 in the same plot, I see that the labels on the left are colored according to the color of each curve (that is nice). However, those labels are some kind of identification string that is difficult to read, and I wish to adjust that.

Since the only thing I can change is the name of the left and right axes, I decided to put each plot in each axis, and then change the labels by hand. But if I do that, both labels are now in black, and I cannot tell which curve is which:

[image]

Is this intended, or is it something that I failed to configure correctly?

Thanks in advance,

ra3xdh, 2022-10-31:
It is not a bug. This limitation is inherited from the Qt4 Qucs version. You may define the label only for the whole axis and cannot edit the trace labels. The fix will require the redesign of the diagrams and may cause backward compatibility issues. So, I don't promise a quick fix for the reported problem. But I am planning to redesign an automatic trace labels generation to make the labels more readable.

ra3xdh, 2022-10-31:
See also #178

ra3xdh, 2022-11-28:
The next version of the Qucs-S will contain less redundant info in the trace names. See #178 for screenshot. This will partially resolve the reported problem. The further extension of the trace names definition may result in loss of the backward compatibility and not considered. Closing this ticket.

closed 2022-11-28 by ra3xdh