ra3xdh/qucs_s #179: Legacy Qucs equations versus Nutmeg (ngspice )

tomhajjar, 2022-10-31:
I have 700 Qucs schematics with Equation blocks that have been "ported" over to Qucs-S. The vast majority of Equation blocks worked without converting them to Nutmeg equations compatible with ngspice. Is there a way to know what Qucs functions don't work in Qucs-S(ngspice). Below are some examples. 

[image]

[image]

[image]

ra3xdh, 2022-11-01:
Hello Tom. there is no list of Ngspice-incompatible functions. The documentation os lacking here. Qucs-S converts only node names (like `out.Vt` and `v(out)`), but not functions.  The `dB()` function is implemented in both simulators, but other functions may be not. Writing a converter for functions may be a complicated task. Simple translation of the functions name may be not sufficient here. Instead of writing a converter, I decided to prevent user from insertion a Qucs equation in schematic using Ngspice simulation. That's why I removed a Qucs equation button from the toolbar and replaced it with Nutmeg equation.

tomhajjar, 2022-11-01:
"Qucs-S converts only node names (like out.Vt and v(out)), but not functions."

Converting names is only true for Qucs equations correct? I cannot use Qucs names in Nutmeg Equations. 

Adding to my confusion was both Qucs and ngspice don't list the full and truncated names that are allowed for functions. 

ph and phase are treated the same in ngspice, 

[image]

ra3xdh, 2022-11-01:
> Converting names is only true for Qucs equations correct? 

Qucs-S doesn't apply any preprocessing for Nutmeg equations. They are transfered to the simulator unchanged. Qucs node names must not be used when SPICE kernel is selected. 

> Does Qucs/ngspice list the full or truncated name allowed for every function?

Only `ph` is listed here https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-23827 The `phase` seems to be undocumented but works as expected.

tomhajjar, 2022-11-02:
Holger pointed me in the right direction.  I also found an error in the Qucs-S Help. Qucs-S Help infers "log" is base 10 when it is base e. This error is on two locations in the Qucs-S Help. I haven't dug into Operator differences. 

[image]

[ngspice Function names.zip](https://github.com/ra3xdh/qucs_s/files/9921890/ngspice.Function.names.zip)

ra3xdh, 2023-03-04:
I have added a list of the most frequently used Nutmeg functions to the Qucs-S tutorial. This should prevent similar questions in the future. Closing this issue as resolved.
[image]

closed 2023-03-04 by ra3xdh

tomhajjar, 2023-03-04:
As I stated above, ngspice allows multiple acronyms and usage for the same function and some are the same as used in Qucs Equations. 

Also causing confusion:

"log"="ln" ;natural log
"log10"; Log base 10

page 106 ngspice 39 manual:
Exponential and logarithmic: exp, ln, log, log10 (ln, log with base e, log10 with base10)