ra3xdh/qucs_s #182: Ngspice simulation error "Only DC simulation found in the schematic. It has no effect!"

d1nuc0m, 2022-11-02:
Hi, I'm trying to follow the [QUCS Get Started](https://qucs.github.io/docs/tutorial/getstarted.pdf) for DC simulation (_page 6, DC simulation - A voltage divider_). I know this is QUCS-S and not QUCS, but the [QUCS-S Get started](https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf) only includes AC simulation.

I realized the same scheme, but when I try to run the simulation I get "_Only DC simulation found in the schematic. It has no effect! Add TRAN, AC, or Sweep simulation to proceed. Simulation finished Ngspice error..._".  Here are the files I used: [voltageDivider..zip](https://github.com/ra3xdh/qucs_s/files/9922131/voltageDivider.zip)

I don't want to use a sweep as I'm not interested in exploring different parameters, so what should I do?

ngspice is installed on my system:
```
$ ngspice --version
******
** ngspice-37 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2022, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Mon Jul 11 11:06:26 UTC 2022
******
```

ra3xdh, 2022-11-02:
See also #78

ra3xdh, 2022-11-02:
I have added a TODO list for this issue.

d1nuc0m, 2022-11-02:
> Use _Simulation->Calculate DC bias_ (or press F8) if you need the OP point
Thank you, is this the appropriate method to solve DC circuits?
If I want to generate for example a table, I have to do a sweep or a transient simulation?

tomhajjar, 2022-11-03:
Try this project. 

To learn how use use Qucs-S, try the examples in the share\qucs-s\examples folder.

[Resistive_Voltage_Divider_prj.zip](https://github.com/ra3xdh/qucs_s/files/9925085/Resistive_Voltage_Divider_prj.zip)

ra3xdh, 2022-11-03:
> If I want to generate for example a table, do I have to do a sweep or a transient simulation?

Currently you need to use DC sweep (DC+attached Par.Sweep) to extract the dependency of the operating point voltage on some circuit parameter.

closed 2023-02-03 by ra3xdh