ra3xdh/qucs_s #183: Finding Sprobes examples

0range0range, 2022-11-07:
There is an example of sub-circuit for testing S-parameters in the documentation:

https://qucs-help.readthedocs.io/en/spice4qucs/RF.html

And it is mentioned, that I can find the probes in:

 `The S-parameter probes, templates and examples introduced in this chapter can be found in the Qucs-S examples directory. They are stored as project spice4qucs-helpChapter13_prj.`

I tried to find this project in the /examples in this repository and in qucs, but could not.  Is the "examples directory" mentioned in documentation lies somewhere else?  Did I miss something?

I did get them from the other source. So this is not the main issue anymore.

ra3xdh, 2022-11-07:
Ngspice has added S-parameter simulation since version 37. Qucs-S supports S-parameter simulation directly since version 0.0.24. See #84 for context The approach with emulation of S-parameter analysis with S-probes described in Chapter 13 is not needed anymore. I have excluded related examples because it may give the user a false direction. The Chapter 13 of the Qucs-S manual need to be 100% rewritten. My advice is to update Ngspice to version >=37 and use S-parameter simulation directly. You may need to compile Ngspice manually, but it depends only on GCC compiler.

0range0range, 2022-11-07:
Actually, I had just recent enough versions of both Qucs-S and ngspice and had it under "Simulations" in Qucs-s. I swear I looked through that list 5 times. :facepalm: 

Thanks!

closed 2022-11-07 by 0range0range