ra3xdh/qucs_s #186: Parameter sweep issues

tomhajjar, 2022-11-22:
I'm still seeing issues when doing DC sweeps of tubes and FETs. I'm think the issue is the same for any Parameter sweep using an even number of points.

[image]

[image]

[Tube_6CA7_Push_Pull_prj.zip](https://github.com/ra3xdh/qucs_s/files/10061395/Tube_6CA7_Push_Pull_prj.zip)

ra3xdh, 2022-11-22:
The Ngspice and Qucsator has different meaning of "step" and "intervals count" sweep. This issue is explained here #8

tomhajjar, 2022-11-22:
So Steps is wrong for ngspice and xyce spice but correct for Qucs. Since Qucsator is being depreciated, is it time to make a change or use the selection of simulator to change the definition?

cmat21, 2023-09-26:
It is somewhat confusing to have a difference in DC sweeps using the same settings. Is it planned to correct this difference?
[image]

ra3xdh, 2023-09-26:
The DC sweep using DC+ParSweep was inherited from old Qucs. Qucsator has no DC sweep simulation and uses this combination instead. It also uses different method to calculate step. The same explantion as here: https://github.com/ra3xdh/qucs_s/issues/8 I will think how to provide a fix and don't break the compatibility between simulators. It is not an obvious task.

ra3xdh, 2024-03-04:
Fixed by #615

closed 2024-03-04 by ra3xdh