ra3xdh/qucs_s #187: Crash on drag'n'drop event

ra3xdh, 2022-11-23:
I think you can reproduce this bug. I am using kubuntu 22.04 and drag the component in components tab and not drop to sch but drop again the components tab then click somewhere in the app then app crash with segmentation fault and similarly but I choose spice specific section same step but app crash with bad alloc and core dump

_Originally posted by @dsm in https://github.com/ra3xdh/qucs_s/issues/92#issuecomment-1324244749_

ra3xdh, 2022-11-25:
It seems to be a Qt version dependent issue. I don't see it on Ubuntu 18.04 using Qt5.9.5

dsm, 2022-11-25:
sometimes it gives std::bad_alloc exception sometimes it gives seg fault I'm trying to compile using qt5.15.7 

Warning: Trying to construct an instance of an invalid type, type id: 167772160
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

ra3xdh, 2022-11-25:
I will try to debug it using Qt5.15

dsm, 2022-11-25:
I am trying to analyze valgrind but I didn't make it. Run app with valgrind make it app unusable

dsm, 2022-11-25:
[image]

dsm, 2022-11-26:
any progress ?

ra3xdh, 2022-11-27:
No, I didn't make further investigations on this issue. As a workaround simple don't use drag'n'drop. Click on a device in the left panel, but don't drag it. Then contour image of the device appears on schematic following mouse pointer. And you may insert the device without using drag'n'drop.

tvrusso, 2022-11-27:
For what it's worth, I can reproduce this problem (took me a while, because I didn't understand the nature of the crash described, and don't use drag-and-drop myself).  On FreeBSD with Qt5.15.5.

Crash is very deep in Qt libraries, and I did not find anything in the stack trace that involved Qucs-s code I could look at --- it was all Qt library functions.

```
(gdb) where
#0  thr_kill () at thr_kill.S:3
#1  0x0000000801f40d44 in __raise (s=s@entry=6)
    at /usr/src/lib/libc/gen/raise.c:52
#2  0x0000000801eb9409 in abort () at /usr/src/lib/libc/stdlib/abort.c:67
#3  0x0000000801d4d259 in report_failure (err=<optimized out>, 
    thrown_exception=0x805a9b788) at /usr/src/contrib/libcxxrt/exception.cc:719
#4  0x0000000801a63f6d in qBadAlloc() ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#5  0x0000000801aafb27 in QListData::detach(int) ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#6  0x0000000801bcc362 in ?? () from /usr/local/lib/qt5/libQt5Core.so.5
#7  0x0000000801c4892d in ?? () from /usr/local/lib/qt5/libQt5Core.so.5
#8  0x0000000801c435ee in QMetaType::load(QDataStream&, int, void*) ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#9  0x0000000801c6e2d8 in QVariant::load(QDataStream&) ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#10 0x0000000801c6d274 in operator>>(QDataStream&, QVariant&) ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#11 0x0000000801bec426 in ?? () from /usr/local/lib/qt5/libQt5Core.so.5
#12 0x0000000801be2e8c in QAbstractItemModel::decodeData(int, int, QModelIndex const&, QDataStream&) () from /usr/local/lib/qt5/libQt5Core.so.5
#13 0x0000000801be843a in QAbstractListModel::dropMimeData(QMimeData const*, Qt::DropAction, int, int, QModelIndex const&) ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#14 0x0000000800f4e74c in QListWidget::dropMimeData(int, QMimeData const*, Qt::DropAction) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#15 0x0000000800f02c02 in QAbstractItemView::dropEvent(QDropEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#16 0x0000000800f388f0 in QListView::dropEvent(QDropEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#17 0x0000000800cd5290 in QWidget::event(QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#18 0x0000000800d7437b in QFrame::event(QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#19 0x0000000800f005cd in QAbstractItemView::viewportEvent(QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#20 0x0000000801c2fc39 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#21 0x0000000800c9bac4 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Widgets.so.5
#22 0x0000000800c9dcc9 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#23 0x0000000801c2f8e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Core.so.5
#24 0x0000000800cf4638 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#25 0x0000000800cf189c in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#26 0x0000000800c9bad9 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Widgets.so.5
#27 0x0000000800c9ceb0 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#28 0x0000000801c2f8e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Core.so.5
#29 0x00000008012ad8ca in QGuiApplicationPrivate::processDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>) () from /usr/local/lib/qt5/libQt5Gui.so.5
#30 0x000000080128c1ea in QWindowSystemInterface::handleDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>) () from /usr/local/lib/qt5/libQt5Gui.so.5
#31 0x00000008039554ee in ?? () from /usr/local/lib/qt5/libQt5XcbQpa.so.5
#32 0x0000000803955100 in ?? () from /usr/local/lib/qt5/libQt5XcbQpa.so.5
#33 0x00000008012ea3f4 in QBasicDrag::eventFilter(QObject*, QEvent*) ()
   from /usr/local/lib/qt5/libQt5Gui.so.5
#34 0x0000000801c2fb01 in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#35 0x0000000800c9ba14 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Widgets.so.5
#36 0x0000000800c9ceb0 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#37 0x0000000801c2f8e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/lib/qt5/libQt5Core.so.5
#38 0x00000008012a8c10 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/local/lib/qt5/libQt5Gui.so.5
#39 0x000000080128d18c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Gui.so.5
#40 0x0000000803942b8f in ?? () from /usr/local/lib/qt5/libQt5XcbQpa.so.5
[... 32 lines cut ...]
    () from /usr/local/lib/qt5/libQt5Core.so.5
#60 0x0000000800c9c44c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from /usr/local/lib/qt5/libQt5Widgets.so.5
#61 0x0000000800cf2719 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#62 0x0000000800cf1661 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
[... cut ...]
#70 0x0000000802b91f34 in ?? () from /usr/local/lib/libglib-2.0.so.0
#71 0x0000000802b91ff8 in g_main_context_iteration ()
   from /usr/local/lib/libglib-2.0.so.0
#72 0x0000000801c84ec0 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Core.so.5
#73 0x0000000801c2c1f6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Core.so.5
#74 0x0000000801c2ff7e in QCoreApplication::exec() ()
   from /usr/local/lib/qt5/libQt5Core.so.5
#75 0x00000000003689ac in main ()
```

tvrusso, 2022-11-27:
Yeah, I'm not getting any information out of gdb even from a debug build.  There are still all of those things in the stack trace with "??" in them that aren't traceable from anywhere.  

I was, however, able to run the debug version with valgrind (it wasn't "usable" but I was actually able to get it to run and reproduce the error with it).

There may be a smoking gun in here, which occurred while I was dragging around a capacitor symbol from the palette to the drawing frame and back.  I do actually get some information about the qucs-s code that is involved, including Schematic::contentsDragLeaveEvent.
```
Debug: pressed CompComps id 2 "Capacitor"
Debug: slotSelectComponent() 0 2
Debug: mod->info true
Debug: mod->infoVA false
Warning: QVariant::save: unable to save type 'iconCompInfoStruct' (type id: 1065).

Debug: pressed CompComps id 2 "Capacitor"
Debug: slotSelectComponent() 0 2
Debug: mod->info true
Debug: mod->infoVA false
Warning: QWidget::paintEngine: Should no longer be called
==33307== Invalid read of size 1
==33307==    at 0x485A21E: memcpy (in /usr/local/libexec/valgrind/vgpreload_memcheck-amd64-freebsd.so)
==33307==    by 0x5E6BC04: ??? (in /lib/libc.so.7)
==33307==    by 0x5E0BA00: ??? (in /lib/libc.so.7)
==33307==    by 0x5E0804A: ??? (in /lib/libc.so.7)
==33307==    by 0x5E07EA6: vfprintf_l (in /lib/libc.so.7)
==33307==    by 0x5E0DF72: fprintf (in /lib/libc.so.7)
==33307==    by 0x3677C2: qucsMessageOutput(QtMsgType, QMessageLogContext const&, QString const&) (qucs/main.cpp:304)
==33307==    by 0x589DD2B: ??? (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x58A2289: QMessageLogger::warning(char const*, ...) const (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x5454B20: QPainter::begin(QPaintDevice*) (in /usr/local/lib/qt5/libQt5Gui.so.5.15.5)
==33307==    by 0x380AEC: Schematic::contentsDragLeaveEvent(QDragLeaveEvent*) (qucs/schematic.cpp:2174)
==33307==    by 0x8CB0E0: Q3ScrollView::viewportDragLeaveEvent(QDragLeaveEvent*) (qucs/qt3_compat/q3scrollview.cpp:1849)
==33307==  Address 0xb36693a is 26 bytes inside a block of size 202 free'd
==33307==    at 0x48557EC: free (in /usr/local/libexec/valgrind/vgpreload_memcheck-amd64-freebsd.so)
==33307==    by 0x35ADAE: QTypedArrayData<char>::deallocate(QArrayData*) (qarraydata.h:240)
==33307==    by 0x359C3F: QByteArray::~QByteArray() (qbytearray.h:495)
==33307==    by 0x36773E: qucsMessageOutput(QtMsgType, QMessageLogContext const&, QString const&) (qucs/main.cpp:298)
==33307==    by 0x589DD2B: ??? (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x58A2289: QMessageLogger::warning(char const*, ...) const (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x5454B20: QPainter::begin(QPaintDevice*) (in /usr/local/lib/qt5/libQt5Gui.so.5.15.5)
==33307==    by 0x380AEC: Schematic::contentsDragLeaveEvent(QDragLeaveEvent*) (qucs/schematic.cpp:2174)
==33307==    by 0x8CB0E0: Q3ScrollView::viewportDragLeaveEvent(QDragLeaveEvent*) (qucs/qt3_compat/q3scrollview.cpp:1849)
==33307==    by 0x8C9E41: Q3ScrollView::eventFilter(QObject*, QEvent*) (qucs/qt3_compat/q3scrollview.cpp:1520)
==33307==    by 0x5A66C38: QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x4BC9AC3: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/local/lib/qt5/libQt5Widgets.so.5.15.5)
==33307==  Block was alloc'd at
==33307==    at 0x4853B74: malloc (in /usr/local/libexec/valgrind/vgpreload_memcheck-amd64-freebsd.so)
==33307==    by 0x58B4102: QArrayData::allocate(unsigned long, unsigned long, unsigned long, QFlags<QArrayData::AllocationOption>) (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x59000CF: QByteArray::QByteArray(int, Qt::Initialization) (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x5ABE54D: ??? (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x5937948: QString::toUtf8_helper(QString const&) (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x36F9B2: QString::toUtf8() const & (qstring.h:684)
==33307==    by 0x367723: qucsMessageOutput(QtMsgType, QMessageLogContext const&, QString const&) (qucs/main.cpp:298)
==33307==    by 0x589DD2B: ??? (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x58A2289: QMessageLogger::warning(char const*, ...) const (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x5454B20: QPainter::begin(QPaintDevice*) (in /usr/local/lib/qt5/libQt5Gui.so.5.15.5)
==33307==    by 0x380AEC: Schematic::contentsDragLeaveEvent(QDragLeaveEvent*) (qucs/schematic.cpp:2174)
==33307==    by 0x8CB0E0: Q3ScrollView::viewportDragLeaveEvent(QDragLeaveEvent*) (qucs/qt3_compat/q3scrollview.cpp:1849)
==33307== 
```

And when I released, it's trying to malloc "-1" somewhere in the course of manipulating QList data.

```
==33307== 
Warning: QPainter::begin: Paint device returned engine == 0, type: 1
Warning: QVariant::load: unable to load type 1065.
Warning: Trying to construct an instance of an invalid type, type id: 788558336
Warning: Trying to construct an instance of an invalid type, type id: 4521984
Warning: Trying to construct an instance of an invalid type, type id: 5242985
Warning: Trying to construct an instance of an invalid type, type id: 553676800
Warning: Trying to construct an instance of an invalid type, type id: 7209029
Warning: Trying to construct an instance of an invalid type, type id: 687894016
[... cut ...]
==33307==    by 0x5A1F439: QAbstractListModel::dropMimeData(QMimeData const*, Qt::DropAction, int, int, QModelIndex const&) (in /usr/local/lib/qt5/libQt5Core.so.5.15.5)
==33307==    by 0x4E7C74B: QListWidget::dropMimeData(int, QMimeData const*, Qt::DropAction) (in /usr/local/lib/qt5/libQt5Widgets.so.5.15.5)
==33307==    by 0x4E30C01: QAbstractItemView::dropEvent(QDropEvent*) (in /usr/local/lib/qt5/libQt5Widgets.so.5.15.5)
==33307== 
==33307== 
==33307== Process terminating with default action of signal 6 (SIGABRT): dumping core
```

But at that point I can get no further information and have to leave it there.

ra3xdh, 2022-11-27:
I compiled the Qucs-S with Qt5.15. I didn't find the root of the crash too. But the setting `setAcceptDrops(false)` for the item list widget helps to supress the crash.

ra3xdh, 2022-11-27:
I have added fix for this on the `current` branch.

ra3xdh, 2022-11-27:
The problem with movable devices icons in the left panel is also gone after the fix.

ra3xdh, 2022-12-03:
Closing this as resolved.

closed 2022-12-03 by ra3xdh

dsm, 2024-11-22:
I sent a PR #1086 to fix this issue.