ra3xdh/qucs_s #188: Qucsator not working

tomhajjar, 2022-11-23:
I haven't used qucsator.exe with Qucs-S in a very long time. I tried today and cannot get it to work. I tried Qucs-S 0.0.24 and 1.0.0 with qucsator.exe from Qucs 0.0.19 and 0.0.20. 

qucsator.exe in Qucs 0.0.19 is 99K
qucsator.exe in Qucs 0.0.20 is 566K, over 5x larger.

Is there something obvious I did wrong? 

[image]
[image]

[2N3055-switch.zip](https://github.com/ra3xdh/qucs_s/files/10078540/2N3055-switch.zip)

tomhajjar, 2022-11-24:
It works if Qucs is installed and change the setting to point to qucsator.exe.

[image]

ra3xdh, 2022-11-24:
Yes, this is expected behavior. The reasons are explained above. You cannot move `qucsator.exe` elsewhere without providing a matching DLLs version. The DLLs from Qucs-S distribution has the same name, but are different from Qt4 Qucs distribution.

tomhajjar, 2022-11-24:
Qucs development is diverging yet again. One group is pushing Gnucsator but they are providing no assistance to Windows users to get it. In addition there are no details of the shortcomings of using Gnucsator.

ra3xdh, 2022-11-24:
gnuscator is a wrapper for Gnucap simulation kernel that provides Qucsator-compatible syntax. It could be used as the direct replacement for qucsator engine. I didn't try gnuscator with Qucs-S but in principle it should operate if one specify executable name pointing to gnucsator. Gnuscator is also not actively developed. I see long (2-3 months) pauses between the commits.

ra3xdh, 2022-11-25:
I assume the issue with qucsator launch on Windows platform has been resolved and the ticket could be closed.

closed 2022-11-25 by ra3xdh